RussTedrake/underactuated #29: Time derivative of energy in a pendulum

robz, 2016-04-26:
The pendulum example in section 11.1 states: 

> Evaluating the time derivative of the energy reveals: 
> dE/dt = −b(θ')^2

My apologies if it's obvious, but where did this come from? If energy of the simple pendulum as a function of time was derived elsewhere in the book, maybe it would be good to add a reference to it from this example.

shensquared, 2016-04-26:
It's from algebra. 

\dot{E}
=ml^2\dot{\theta}\ddot{\theta}+\dot{\theta}mgl\sin{\theta}
=\dot{\theta}*(ml^2\ddot{\theta}+mgl\sin{\theta})
=\dot{\theta}*(-b\dot{\theta})

The last equality is from plugging in the motion equation (the first equation in the example). This example serves more as an introduction to the Lyapunov analysis intuition; Chapter 3 talks about the energy shaping controllers more specifically.

robz, 2016-05-08:
Ah, that makes sense. Thanks @shensquared !

closed 2016-05-08 by robz