RussTedrake/underactuated #59: matplotlib visualizer complaints

peteflorence, 2018-02-05:
Not sure where visualizer development is at but getting these:

```
/usr/lib/python2.7/dist-packages/matplotlib/figure.py:397: UserWarning: matplotlib is currently using a non-GUI backend, so cannot show the figure
  "matplotlib is currently using a non-GUI backend, "
```

Running the block of code twice makes the warning go away, and displays a static image

Quick look around suggests there are fixes for this warning: https://github.com/mapbox/rasterio/issues/686

RussTedrake, 2018-02-05:
how were you running it?  (from the console?  from a notebook?  which platform?)

peteflorence, 2018-02-05:
Same complaints in these two configurations (as expected since both are
Docker):

- Locally running docker container on Mac, locally viewing ipython notebook
in local browser
- Remote running docker container on Linux, remote viewing ipython notebook
in local browser

RussTedrake, 2020-01-19:
Closing this for now.  Will watch for it to rear it's head again with our updated workflows (python3 , colab, etc)

closed 2020-01-19 by RussTedrake

RussTedrake, 2020-02-01:
fwiw -- I think i've resolved this in https://github.com/RobotLocomotion/drake/pull/12658