RussTedrake/underactuated #63: Issues during native Mac install via binaries

peteflorence, 2018-02-05:
Would have just made PR for first issue but not sure if this was particular to something on my system

Following instructions [here](http://underactuated.csail.mit.edu/underactuated.html?chapter=drake)

### Issue 1
Needed this:
```
export PYTHONPATH=/opt/drake/lib/python2.7/site-packages:${PYTHONPATH}
```
instead of
```
export PYTHONPATH=/opt/drake/install/lib/python2.7/site-packages:${PYTHONPATH}
```

(don't have a `/opt/drake/install`)

### Issue 2

```
Petes-MacBook-Pro:russdrake pflomacpro$ python -c 'import pydrake; print(pydrake.__file__)'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/drake/lib/python2.7/site-packages/pydrake/__init__.py", line 8, in <module>
    from . import common
  File "/opt/drake/lib/python2.7/site-packages/pydrake/common.py", line 3, in <module>
    from ._common_py import *
ImportError: dlopen(/opt/drake/lib/python2.7/site-packages/pydrake/_common_py.so, 2): Library not loaded: /usr/local/opt/vtk@8.0/lib/libvtkFiltersSources-8.0.1.dylib
  Referenced from: /opt/drake/lib/python2.7/site-packages/pydrake/_common_py.so
  Reason: image not found
```
Looks like vtk issues (maybe is finding a different vtk I have?)

peteflorence, 2018-02-05:
Russ fwded me that the PYTHONPATH is a known typo and will be fixed

I would vote not to worry too much about Issue 2 though -- I'd prefer to just focus on supporting dockered drake.

Also I think in addition to dockered drake via binaries, the other configurations I'd vote to focus on supporting are configurations building from source that allow students to contribute back to Drake:

1. Dockered drake distribution geared for class where students can build from source inside
2. Native Linux install building from source
3. Native Mac install building from source

@gizatt too

gizatt, 2018-02-05:
Absolutely! Though aren't 2) and 3) functionally free, as that just entails following the Drake build instructions? 1) we could provide pretty simply.

I don't want to have either us or the students distracted supporting those, though.

peteflorence, 2018-02-05:
That's right that 2 and 3 are proper Drake build instructions :)  yes would hope students go that route if they are interested.

True that providing 1, for example via images we push to `gizatt/drake_6832:drake-20180203-source` instead of `gizatt/drake_6832:drake-20180203` would be non-zero extra work (probably want to mount drake src externally like we do in Spartan... etc).  Perhaps best to just let students try building from source via 2 or 3, since those are great options.

peteflorence, 2018-02-08:
The PYTHONPATH has since been fixed. Not sure about the Mac binary install but here isn't the best place to track it

closed 2018-02-08 by peteflorence