RussTedrake/underactuated #83: simple_continuous_time_system fails to build on mac

RussTedrake, 2018-02-14:
@jamiesnape 

I've verified that this is also true on shambhala/drake_cmake_installed (I get the same errors):

```
[ 50%] Building CXX object src/CMakeFiles/simple_continuous_time_system.dir/simple_continuous_time_system.cc.o
In file included from /Users/russt/underactuated/src/simple_continuous_time_system.cc:9:
In file included from /Users/russt/drake-build/install/include/drake/systems/analysis/simulator.h:3:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:640:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:635:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/iterator:417:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/iosfwd:90:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/wchar.h:119:
In file included from /usr/include/wchar.h:91:
In file included from /Users/russt/drake-build/install/include/fmt/time.h:13:
In file included from /Users/russt/drake-build/install/include/fmt/format.h:38:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stdexcept:83:32: error: unknown type name 'string'
    explicit logic_error(const string&);
                               ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stdexcept:100:34: error: unknown type name 'string'
    explicit runtime_error(const string&);
                                 ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stdexcept:115:59: error: unknown type name 'string'
    _LIBCPP_INLINE_VISIBILITY explicit domain_error(const string& __s) : logic_error(__s) {}
                                                          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stdexcept:125:63: error: unknown type name 'string'
    _LIBCPP_INLINE_VISIBILITY explicit invalid_argument(const string& __s) : logic_error(__s) {}
                                                              ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stdexcept:135:59: error: unknown type name 'string'
    _LIBCPP_INLINE_VISIBILITY explicit length_error(const string& __s) : logic_error(__s) {}
                                                          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stdexcept:145:59: error: unknown type name 'string'
    _LIBCPP_INLINE_VISIBILITY explicit out_of_range(const string& __s) : logic_error(__s) {}
                                                          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stdexcept:155:58: error: unknown type name 'string'
    _LIBCPP_INLINE_VISIBILITY explicit range_error(const string& __s) : runtime_error(__s) {}
                                                         ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stdexcept:165:61: error: unknown type name 'string'
    _LIBCPP_INLINE_VISIBILITY explicit overflow_error(const string& __s) : runtime_error(__s) {}
                                                            ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stdexcept:175:62: error: unknown type name 'string'
    _LIBCPP_INLINE_VISIBILITY explicit underflow_error(const string& __s) : runtime_error(__s) {}
                                                             ^
```

jamiesnape, 2018-02-14:
Is this on the same system that we discussed yesterday?

RussTedrake, 2018-02-14:
No.  Completely separate.

Yesterday, my work laptop failed to dynamically load pydrake from a local build/install the python interpreter.

Today, on my home machine (which loads pydrake just fine), i still cannot build this cc example in underactuated nor shambhala.

jamiesnape, 2018-02-14:
Builds fine for me using the latest experimental mac build.
```
$ git rev-parse HEAD
b3a87c6d5f51b9ba1dd50ac4fe62189e9a38fe68
$ xcodebuild -version
Xcode 9.2
Build version 9C40b
$ xcode-select -p
/Applications/Xcode.app/Contents/Developer
$ cat drake/share/doc/drake/VERSION.TXT 
20180214092416 a6d1c905f390300cb20313721e37c659d587a75f
$ cmake -Ddrake_DIR=drake ..
-- The C compiler identification is AppleClang 9.0.0.9000039
-- The CXX compiler identification is AppleClang 9.0.0.9000039
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/local/bin/python2.7 (found suitable version "2.7.14", minimum required is "2.7") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Found Protobuf: /usr/local/lib/libprotobuf.dylib (found suitable version "3.5.1", minimum required is "2.6.1") 
-- Found PythonInterp: /usr/local/bin/python2.7 (found version "2.7.14") 
-- Configuring done
-- Generating done
-- Build files have been written to: .
hippo:build snape (git: master)$ make
Scanning dependencies of target simple_continuous_time_system
[ 50%] Building CXX object src/CMakeFiles/simple_continuous_time_system.dir/simple_continuous_time_system.cc.o
[100%] Linking CXX executable simple_continuous_time_system
[100%] Built target simple_continuous_time_system
```

closed 2019-03-30 by RussTedrake