RussTedrake/underactuated #85: experimental mac and ubuntu builds have different timestamps

RussTedrake, 2018-02-14:
my current drake_version.json logic does not support that.  so, as it stands, the links to the mac binaries in the text will be broken whenever drake_version is pointing to the (linux) experimental.

fyi @jamiesnape .  (but unless there is a trivially different naming solution, i think this one is on me to fix).

jamiesnape, 2018-02-14:
I guess the time part of the timestamp is not essential to disambiguate (we have the commit SHA, and really there is only very limited need to keep multiple packages built with the same SHA on the same day), though it is useful for debugging. Ideally, I think we would keep it, but you are the main consumer of these packages, so we should change if it makes life easier.

RussTedrake, 2018-02-14:
It’s only a few more lines of code for me to get it handled on my side.  No worries.

commit 17f43594 (2018-03-11): support separate mac version timestamp.  resolves #85
closed 2018-03-11 by RussTedrake