RussTedrake/underactuated #145: Question about ../src/lyapunov/cubic_polynomial.py

Jingwen-Zhang-Aaron, 2018-09-24:
Hello,

Can anyone help me explain the code of one line inside ../src/lyapunov/cubic_polynomial.py

"prog.AddSosConstraint((V-rho) * x.dot(x) - lambda_ToExpression() * Vdot) "

Why the SOS constraint is like above?

I feel like it should be like  " -Vdot - lambda_ToExpression() * ( rho - V ) ". But it will cause some bugs when I tried this expression.

Can you provide more info about how to use SOS optimization to get the ROA?

Thanks!

Jingwen-Zhang-Aaron, 2018-09-26:
Coool! I just read your online book and found that the code cannot match with what you have written in the chapter so that I got lost a little bit here. Thank you so much for your explanation. What a clever way!

Jingwen-Zhang-Aaron, 2018-09-26:
Hello, I wanna use SOS to do the lyapunov analysis of the simple pendulum. There's no bug for my code but I got V = 0 and lambda = 0 which indicates obviously that something goes wrong. Can you help me have a look at my code? 
I just followed the following formulation to find the V.
[image]

```python
import math
import numpy as np

from pydrake.all import (Jacobian, MathematicalProgram, SolutionResult,
                         Variables)

def dynamics(x):
    return [x[1]*x[2], -x[0]*x[2], -(x[2] + 9.81*x[0])]

prog = MathematicalProgram()
x = prog.NewIndeterminates(3, "x")

# Define the Lyapunov function.
(V, constraint1) = prog.NewSosPolynomial(Variables(x), 2)
Vdot = Jacobian([V.ToExpression()], x).dot(dynamics(x))[0]

# Define the Lagrange multipliers.
(lambda_, constraint2) = prog.NewSosPolynomial(Variables(x), 4)

prog.AddSosConstraint(-Vdot - lambda_.ToExpression()*(pow(x[0],2) + pow(x[1],2) - 1))

result = prog.Solve()

assert(result == SolutionResult.kSolutionFound)

print(prog.SubstituteSolution(V.ToExpression()))
print(prog.SubstituteSolution(lambda_.ToExpression()))

```

RussTedrake, 2019-03-30:
my version of this is here: https://github.com/RussTedrake/underactuated/blob/master/src/pendulum/global_sums_of_squares.py
i didn't check every line, but notice that i was careful to set the scaling by setting V(1) = 1.  Otherwise V=0 is perfectly viable.

closed 2019-03-30 by RussTedrake