RussTedrake/underactuated #155: Consider a notation change to support output feedback

RussTedrake, 2019-01-03:
Throughout the notes I use `xdot = f(x,u)` for the state-space dynamics, and `g(x,u)` for the running cost in optimal control.  And then the few times I use output so far in the notes, I use `y=h(x,u)`.  This is not so uncommon (https://en.wikipedia.org/wiki/State-space_representation#Nonlinear_systems, https://en.wikipedia.org/wiki/Observability#Nonlinear_case).

But, it would be much more satisfying to have `xdot = f(x,u), y = g(x,u)`.  I could switch to e.g. `c(x,u)` for my running cost -- but would have to touch a lot of places in the text.

RussTedrake, 2020-01-14:
@TobiaMarcucci -- any opinions on this one?

TobiaMarcucci, 2020-01-14:
As you say, using `f(x,u)` for the dynamics and `h(x,u)` for the output is not unusual. For example Khalil does that in "Nonlinear systems" and Isidori does that in "Nonlinear control systems". However, they both do that to save `g(x)` for the coefficient of `u` in the dynamics (i.e. `xdot = f(x,u) + g(x)u`). So I agree that the current solution is not super clean. I'd also go for `xdot = f(x,u)`, `y = g(x,u)` as long as control-affine dynamics do not appear too often.

Regarding the running cost in the OCPs. In dynamic programming, e.g. Bertsekas' book, I've seen `g(x,u)` as you are using right now. However, in most of the modern "analytical optimal control" books I've read, the integrand term is `l(x,u)` or `L(x,u)` (e.g., Liberzon's and Bressan and Piccoli's). Could `l(x,u)` be better than `c(x,u)`, provided that no other conflicts arise?

RussTedrake, 2020-01-16:
thanks!  I've done xdot = f1(x,u) + f2(x)u for control affine in the intro, so that overlap is not a big concern.

yes, I could be happy, I think, with l(x,u) for the running cost.

commit 5cae9c20 (2020-01-20): change notation g(x,u) => l(x,u) to support output feedback using g(x,u) as the output.  Resolves #155.
closed 2020-01-20 by RussTedrake