RussTedrake/underactuated #158: Cable-driven robots can be partially underactuated

aarchiba, 2019-01-30:
The simplest example is a one-dimensional cable-driven robot actuated by a single motor that winds a cable the robot hangs from. If the actuator is falling at 1g the motor has at best unidirectional control over the motion. Of course this robot is fairly trivial, but people are building two-dimensional graffiti bots that hang from two cables and can move over a building wall; they suffer the same sort of constraint, and motion planning is nontrivial. More common is to overconstrain such robots; for example to use eight cables for a six-axis robot. I believe this is what is used for coarse positioning of the receiver room of the Chinese radio telescope FAST. These overconstrained systems always operate near but not in the going-slack region (which has a complicated shape), and many of them (okay not FAST) are interesting for rapid motion, since no motors need be moved.

peteflorence, 2019-01-30:
Is there somewhere in Russ’s text that says cable-driven robots can never be underactuated?

aarchiba, 2019-01-30:
No, this was in response to: "Admittedly, this permits the existence of a gray area, where we would prefer to avoid describing the system as either fully- or underactuated (we should instead only describe its states); but if you find a physical system that is sort of "half and half", then do let me know!"

peteflorence, 2019-01-30:
:) 👍

PR #159 (2019-02-02): address the gray area.  resolves #158.
Resolves #158
commit 18113034 (2019-02-02): address the gray area.  resolves #158. (#159)
closed 2019-02-02 by RussTedrake