RussTedrake/underactuated #169: acrobot dircol example fails with --config snopt_fortran, but succeeds with the precompiled binaries

RussTedrake, 2019-02-24:
setting num_time_samples=25 resolves it, but at the cost of a much less optimal trajectory coming out.

cc @hongkai-dai, @jwnimmer-tri @jamiesnape .  I guess we're not using the fortran snopt in the precompiled binaries yet?

RussTedrake, 2019-02-24:
i've confirmed that changing my bazel settings on the local install from `--config=snopt_fortran` back to `--config=snopt` fixes the issue on this machine.  this confirms for me that it is a difference in snopt between versions.

jwnimmer-tri, 2019-02-25:
> I guess we're not using the fortran snopt in the precompiled binaries yet?

That is correct; RobotLocomotion/drake#10422 is the overall tracking issue.

> ... it is a difference in snopt between versions.

In your builds that failed/passed when toggling `--config=snopt_fortran`/`--config=snopt`, which source release of SNOPT were you testing the acrobot dircol against?  Assuming you were using Bazel builds, the question is what was `env SNOPT_PATH` set to?
 
We're going to need to categorize this failure as either one of the "currently known bugs in SNOPT Fortran" shown in RobotLocomotion/drake#10422, or else file a new issue for this failure mode if its different.  It's possible that @hongkai-dai has some insight in that regard, or might be able to suggest some tolerance or tuning options that can mitigate this change.

RussTedrake, 2019-02-25:
i have been testing on CI against the "latest" "continuous" drake-binaries
https://github.com/RussTedrake/underactuated/blob/master/drake_version.json

I also run nightly tests against "nightly" (I pick early AM to be after the drake nightlies come out).  have not seen the failure.   My CI does not set SNOPT_PATH.  https://github.com/RussTedrake/underactuated/blob/master/.circleci/config.yml

On my machine (where I was seeing failures when I flipped to snopt_fortran), I build directly from master, and do set SNOPT_PATH="git".

jwnimmer-tri, 2019-02-25:
> ... testing on CI against the "latest" "continuous" drake-binaries; also ... against "nightly"; have not seen the failure. My CI does not set SNOPT_PATH.

With precompiled Drake, you are getting SNOPT 7.4 source archive built using F2C mode.

> On my machine (where I was seeing failures when I flipped to snopt_fortran), I build directly from master, and do set SNOPT_PATH="git".

That's the SNOPT 7.4 source archive (as of RobotLocomotion/drake#10430) built using Fortran mode.

Given that data, it sounds to me like this issue is neither RobotLocomotion/drake#10423 (unbounded) nor RobotLocomotion/drake#10424 (bug in 7.6 fortran sources, not 7.4 fortran sources).

RussTedrake, 2019-02-25:
To be clear, I don't think we can declare this new instance as a bug.  The nonlinear optimization is not guaranteed to find a solution to this problem.  But it is a degradation in performance.  7.4 seems to solve it well, and 7.6 does not.

jwnimmer-tri, 2019-02-25:
> ... and 7.6 does not.

nit: The above data only shows that 7.4/Fortran does not solve it well (but 7.4/F2C does).  We haven't tested 7.6/Fortran yet.

RussTedrake, 2020-01-13:
I believe this has been resolved, as since https://github.com/RobotLocomotion/drake/issues/10422 has been completed. will reopen if more testing results in a different understanding.

closed 2020-01-13 by RussTedrake