RussTedrake/underactuated #173: drake's meshcat bump seems to have broken this repo

RussTedrake, 2019-03-03:
https://circleci.com/gh/RussTedrake/underactuated/981

@gizatt -- any ideas?

gizatt, 2019-03-03:
Yeah same thing's happening on mine. https://travis-ci.org/gizatt/pydrake_kuka/builds/500933698

I'm mostly guessing here from quickly looking around, but it seems likely related to [this tweak to meshcat's prereqs](https://github.com/rdeits/meshcat-python/commit/b580c1293b0229c0a0909adff9bdeef934a5cf4d#diff-2eeaed663bd0d25b7e608891384b7298), which includes some conditionals to prefer tornado < 6 when Python 2 is still being used, because Tornado 6 dropped Python < 3 support. Maybe our build servers are still grabbing Tornado 6 for some reason (it went live very recently)?

gizatt, 2019-03-03:
It even happened on my [periscope tutorial](https://travis-ci.org/gizatt/drake_periscope_tutorial/builds/500660171), which builds against some pegged old Drake binaries that use the old meshcat.

gizatt, 2019-03-03:
Using [this build log](https://api.travis-ci.org/v3/job/500962092/log.txt) as a reference, the chain of events leading to Tornado >= 6.0 being installed seems to be the underactuated repo's install-prereqs bringing in `ipykernel`, which brings in `tornado` as a prereq and grabs the most recent version. Opened a PR to test that

closed 2019-03-30 by RussTedrake