RussTedrake/underactuated #207: make all src colab friendly

RussTedrake, 2019-04-28:
Currently we occasionally reference local files, etc, that don't work on colab.

Also need to implement the jupyter widget version of the slider demos.

RussTedrake, 2019-05-15:
https://stackoverflow.com/questions/55622543/colab-scratchpad-nameerror-name-underactuated-is-not-defined

EricCousineau-TRI, 2019-06-12:
If it's all helpful, I can post code in Anzu that we have for Jupyter slider version of tree manipulation? (Though this goes in a bit different direction, and handles each instances separately, and doesn't currently permit manipulation of free-floating bodies...)

RussTedrake, 2019-06-12:
super helpful.  would it make sense to even PR it to drake?  (like the simple UI joint sliders?)

EricCousineau-TRI, 2019-06-12:
Hm... would prefer not to PR it to Drake until we address some part of this guy:
https://github.com/RobotLocomotion/drake/issues/9650

That being said, maybe I can try to do that a bit more in Q3, in such a way that it's only supported on Bionic, and maybe only Python 3.

For the time being, though, here's the code as a Gist:
https://gist.github.com/EricCousineau-TRI/9091c8f51df368d49ab81b1c885b11e4

RussTedrake, 2020-01-20:
I've fixed the local file references problem, but still need to think about how the sliders will work.  Presumably i won't ever get live updates to the `PlanarSceneGraphVisualizer` outputs in colab... perhaps I can make meshcat work?

RussTedrake, 2020-01-20:
using argparse also seems to take down colab.
the following code crashes the interpreter when executing the last line.
```
import argparse
parser = argparse.ArgumentParser()
parser.add_argument("-T", "--duration",
                    type=float,
                    help="Duration to run sim.",
                    default=10000.0)
args = parser.parse_args()
```

EricCousineau-TRI, 2020-01-21:
Can I ask why you're using `argparse` in Jupyter notebooks?

> [...] but still need to think about how the sliders will work [...]

I can send you a link to Jupyter-based sliders in Anzu; I could make a PR to Drake with it, but I may not be able to do that until next week or later.

> Presumably i won't ever get live updates to the PlanarSceneGraphVisualizer [...]

If you use `%matplotlib notebook`, I think you *could* get live updates; I believe the GUI updates from widget sliders would be asynchronous while running the sim, so as long as `AdvanceTo()` is run in certain increments (like [Evan's suggestion](https://github.com/RobotLocomotion/drake/issues/10497#issuecomment-457603106)) that allow Python to dispatch other threading stuff, then it might work?

RussTedrake, 2020-01-21:
I’m not using argparse in the notebooks, but have my examples in .py files with a link to the colab “scratchpad” with the suggestion that one could copy and paste any py file in and run.

EricCousineau-TRI, 2020-01-21:
Ah, gotcha. Some options:

(a) update the interfaces to decompose the `args` output via `main()`, thus the rest of the code can be copy-and-pasted without that part,
(b) replace `args = parser.parse_args()` with something like `args = argparse.Namespace(...)` and fill in what you need.
(c) Put a comment next to `parser.parse_args()` to specify argv when using it in the notebook, e.g. `args = parser.parse_args(args=["--example_flag", "--more_flags"])`.

(a) is the cleanest APi-wise, but (c) would be the quickest?

RussTedrake, 2020-02-11:
Update: binder supports matplotib notebook and is the best user experience.
The steps: 
```
ani = visualizer.get_recording_as_animation()
HTML(ani.to_jshtml())
```
are general (work on colab) but are very slow.  luckily, adding `visualizer.start_recording()` doesn't seem to slow things down.

here's what i think I need to do.  
- check if I'm on colab.  https://stackoverflow.com/questions/53581278/test-if-notebook-is-running-on-google-colab  if yes, set backend to inline, if no, set it to notebook.  (possibly with a comment that other options are available)
- in PlanarSceneGraphVisualizer, the default behavior for show should be show=True iff the current backend is interactive.  https://stackoverflow.com/questions/5091993/list-of-all-available-matplotlib-backends
Then, having code that creates the animation in a separate cell (or which runs only if necessary; e.g. non-interactive backend) seems quite reasonable.

that should give people a nice experience on binder, and a sufficient experience on colab, without too much code bloat. (though admittedly there is a little)

RussTedrake, 2020-02-13:
I've now established the workflow that, I think, supports all of this, and used in for the examples in the pendulum chapter.  Will continue to march through the notes making them better, but don't need this issue to track it.

closed 2020-02-13 by RussTedrake