RussTedrake/underactuated #219: Can't import 'PlanarSceneGraphVisualizer' in the python example 1.1 from chapter 1 in the online textbook

pj381, 2019-06-10:
I ran this python example on my own ubuntu 16.04 machine(not a VM). 
I changed directories into the notebooks directory (which also contains drake utility scripts) 
I then types in this into the terminal :

 **$ ./docker_run_notebook.sh drake-20190129 ./** 

Then I changed directories from within the jupyter terminal as follows:
 
**$  cd /.. 
$ cd underactuated/src/** 

I then opened a juptyer notebook (Python2) and plugged in the code from chapter one under python example 1.1,
and the error it threw up read: 

**ImportError: cannot import name PlanarSceneGraphVisualizer** 

So,.. I also tried to work around this by 

1) Importing all of the underactuated module as ua and prepending ' ua. ' to wherever 'PlanarSceneGraphVisualizer' and 'FindResource' were called and 
2)Commenting out the line where 'PlanarSceneGraphVisualizer' and 'FindResource' were imported in the original code 

but these 2 steps also didn't work. 
The output console would point to this line: 

**---> 14 parser.AddModelFromFile(ua.FindResource("double_pendulum/double_pendulum.urdf"))** 

with a runtime error message as follows: 

**RuntimeError: The resulting spatial inertia is not physically valid. See SpatialInertia::IsPhysicallyValid()** 

---------------------------------------------------------------------------
And that's the issue (at least as best as I could explain it).
I'd appreciate any help.
Thanks.

gizatt, 2019-06-10:
Thanks for reporting! The second issue you report is a sign of a mismatch between the underactuated code version and underlying Drake version -- trying changing the Docker line to use `drake-20190508` rather than `drake-20190129`. (Drake got slightly nicer about handling invalid inertias for non-leaf links over the last few months, I believe.)

The first issue you report is a sign of a weird PYTHONPATH keeping you from importing the underactuated course libraries, which is really weird in a Docker environment. I wasn't able to replicate from your instructions, but if you follow the path you used to get the invalid-inertia error, you should find that resolved if you use the more modern Drake container listed above.

pj381, 2019-06-11:
Hi.
Thanks for responding.

I changed the docker line to use drake-20190508 to update the container like you said
But now when I import 'underactuated' as 'ua ' and prepend 'ua.' to 'FindResource' and 'PlanarSceneGraphVisualizer' I don't even get the invalid-inertia error anymore like I used to.
Instead, I get : 

**AttributeError:** 'module' object has no attribute 'PlanarSceneGraphVisualizer'

As shown below:

[image]

Whereas when I run the example as is 
(WITHOUT importing 'underactuated' as 'ua ' 
OR prepending 'ua.' to 'FindResource' and 'PlanarSceneGraphVisualizer') 
I get:

**ImportError:** cannot import name PlanarSceneGraphVisualizer

As shown below:

[image]

-----------------------
P.S. : Just in case you need to know the setup I used 
          Here's a look at the directories and all the commands I used from my local machine:

[image]

 And here's what I did inside the jupyter terminal : 

[image]

Note that :
              
PlanarSceneGraphVisualizer doesn't show up here when I list he files within
 **/underactuated/src/underactuated** .

And also that I did go back up one directory to **underactuated/src**
before opening the notebook to paste code example in it.   

-------------
Thanks again.

pj381, 2019-06-11:
Feel free to ask me to re-do my setup if that's necessary.
Or to suggest any alternative setups where I can run all examples from the textbook.
Also I just went through more of the textbook...and a lot of the examples there are in colab.
(Except maybe this one.)
So if that's better, you could ask me to switch to colab

gizatt, 2019-06-14:
Ah, I see. Looks like the underactuated-class Docker images are using an out-of-date version of this repo. You could try, as a quick stab, updating the underactuated repo files in the docker container. (As in, fire up the container and git-checkout the latest, and maybe run the install prereqs again to make sure they're updated). It looks as is switching to colab is the path of least resistance for you once #218 gets fixed (it's next on my stack).

To avoid a lot of that opaque trusting-other-peoples-containers headache, you could jump to building examples locally. You can follow [these directions] to install Drake binaries and get them on your Python path, and then clone this repo (the underactuated repo) and put `underactuated/src` on your Python path as well. At that point, the examples in this repo should work (modulo running `drake/share/drake/setup/install_prereqs.sh` and `underactuated/scripts/setup/ubuntu/16.04/install_prereqs`).

RussTedrake, 2020-01-19:
I think we understood this one (thanks @gizatt) and that #248 is the right path to a solution.

closed 2020-01-19 by RussTedrake