RussTedrake/underactuated #230: AttributeError: 'Quadrotor2D_[float]' object has no attribute 'DeclareVectorInpu tPort'

vinbo, 2019-09-15:
Hi, when I copy the src/quadrotor2d directory into the src directory of the course drake toolbox via docker, through the notebook directory, I find the scripts cannot be executed. the error output is:

root@c868a13c9742:/underactuated/src/quadrotor2d# python lqr.py
Traceback (most recent call last):
  File "lqr.py", line 37, in <module>
    plant = builder.AddSystem(Quadrotor2D())
  File "/drake/lib/python2.7/site-packages/pydrake/systems/scalar_conversion.py"
, line 188, in system_init
    self, *args, converter=converter, **kwargs)
  File "/underactuated/src/quadrotor2d/quadrotor2d.py", line 22, in _construct
    self.DeclareVectorInputPort("u", BasicVector_[T](2))
AttributeError: 'Quadrotor2D_[float]' object has no attribute 'DeclareVectorInpu
tPort'

so which part is not right? Thanks

RussTedrake, 2019-09-16:
I don't know how you've gotten this particular error, but I've converted these notes over to using python3 only -- that's what is tested on CI.  Perhaps our docker needs updating?

vinbo, 2019-09-16:
> 
> 
> I don't know how you've gotten this particular error, but I've converted these notes over to using python3 only -- that's what is tested on CI. Perhaps our docker needs updating?

I havn't seen why verson of python matters. I find in the docker, python version is 3.6.5, and in the drake toolbox, by command "$ ./docker_run_bash_mingw.sh drake-20190305 . 192.169.99.1", the python version is 2.7.12.  Does this help?

RussTedrake, 2020-01-19:
Again, not sure how to move forward with this.  Hopefully things will get better with colab and the new docker workflow this term.

closed 2020-01-19 by RussTedrake