RussTedrake/underactuated #231: Symmetry matrix not necessarily positive semi definite

vinbo, 2019-09-27:
Dear Prof, in problem set_3, part 3.3 LQR, in the sandbox, the code is
...
def create_lqr(A, B):
    Q = np.zeros((4, 4))
    # Not clear what these gains will do,
    # but as long as Q is positive semidefinite
    # this should find a solution.
    Q = np.random.random((4, 4))
    Q = Q.T + Q # make symmetric and thus psd
...
But  the symmetrization of Q does not make it positive semi definite, only make the eigenvalue real. Please check, thanks

RussTedrake, 2019-09-28:
Thanks for catching that.  I completely agree, that's a mistake.  

@gizatt -- might you be able to fix it?
https://github.com/RobotLocomotion/6-832-backend/commits/master/sets/3/set_3.ipynb

gizatt, 2019-09-30:
Yeah, agreed that this is slightly wrong. IIRC I found it to be surprisingly-often-still-PSD but I'll tidy it up. Thanks @vinbo!

vinbo, 2019-10-01:
No, most of the time it is not psd. It just seemingly does not affect the expected result of riccati equation solving, havn't figured out why.

btw, what is IIRC?

vinbo, 2019-10-01:
No, most of the time it is still not psd. It just seemingly does not affect the expected result of riccati equation solving, havn't figured out why.

btw, what is IIRC?

closed 2019-10-14 by RussTedrake