RussTedrake/underactuated #235: OSError: [Errno 12] Cannot allocate memory

vinbo, 2019-11-14:
I'm doing problem set4, the hopper, after i run the simulator many times, like several hours for adjusting the parameters. the visualizer breaks down. the error info:

---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-75-cf4a5df6faa4> in <module>()
      3 ani = viz.animate(state_log, 30, repeat=True)
      4 plt.close(viz.fig)
----> 5 HTML(ani.to_html5_video())

/usr/local/lib/python2.7/dist-packages/matplotlib/animation.pyc in to_html5_video(self, embed_limit)
   1350                                 bitrate=rcParams['animation.bitrate'],
   1351                                 fps=1000. / self._interval)
-> 1352                 self.save(f.name, writer=writer)
   1353 
   1354             # Now open and base64 encode

/usr/local/lib/python2.7/dist-packages/matplotlib/animation.pyc in save(self, filename, writer, fps, dpi, codec, bitrate, extra_args, metadata, extra_anim, savefig_kwargs)
   1189                           "animation.")
   1190                 rcParams['savefig.bbox'] = None
-> 1191             with writer.saving(self._fig, filename, dpi):
   1192                 for anim in all_anim:
   1193                     # Clear the initial frame

/usr/lib/python2.7/contextlib.pyc in __enter__(self)
     15     def __enter__(self):
     16         try:
---> 17             return self.gen.next()
     18         except StopIteration:
     19             raise RuntimeError("generator didn't yield")

/usr/local/lib/python2.7/dist-packages/matplotlib/animation.pyc in saving(self, fig, outfile, dpi, *args, **kwargs)
    235         '''
    236         # This particular sequence is what contextlib.contextmanager wants
--> 237         self.setup(fig, outfile, dpi, *args, **kwargs)
    238         try:
    239             yield self

/usr/local/lib/python2.7/dist-packages/matplotlib/animation.pyc in setup(self, fig, outfile, dpi)
    349         # Run here so that grab_frame() can write the data to a pipe. This
    350         # eliminates the need for temp files.
--> 351         self._run()
    352 
    353     def _run(self):

/usr/local/lib/python2.7/dist-packages/matplotlib/animation.pyc in _run(self)
    361                                       stdout=output, stderr=output,
    362                                       stdin=subprocess.PIPE,
--> 363                                       creationflags=subprocess_creation_flags)
    364 
    365     def finish(self):

/usr/local/lib/python2.7/dist-packages/subprocess32.pyc in __init__(self, args, bufsize, executable, stdin, stdout, stderr, preexec_fn, close_fds, shell, cwd, env, universal_newlines, startupinfo, creationflags, restore_signals, start_new_session, pass_fds)
    612                                     c2pread, c2pwrite,
    613                                     errread, errwrite,
--> 614                                     restore_signals, start_new_session)
    615             except:
    616                 # The cleanup is performed within the finally block rather

/usr/local/lib/python2.7/dist-packages/subprocess32.pyc in _execute_child(self, args, executable, preexec_fn, close_fds, pass_fds, cwd, env, universal_newlines, startupinfo, creationflags, shell, p2cread, p2cwrite, c2pread, c2pwrite, errread, errwrite, restore_signals, start_new_session)
   1209                                 errread, errwrite,
   1210                                 errpipe_read, errpipe_write,
-> 1211                                 restore_signals, start_new_session, preexec_fn)
   1212                         self._child_created = True
   1213                     else:

OSError: [Errno 12] Cannot allocate memory

ps: my log output is normal, still in stable state.  something like this:
------------flight phase----------------time: 33.5453820229
x= -0.00520459431137   xd= -0.0136928172646
CGprint= -0.00430172541253    xerr= -0.00273856345291    xTD= -0.0103678768158
theta= -1.4595140219e-15    alpha= -0.00489307818757   alpha_desired= 0.00516330544649
Done
I'm in the container of drake-20190423

So, any idea how to fix this? Thanks

RussTedrake, 2019-11-14:
I'm afraid that the docker instances are not going to be updated again until the course runs in the spring, but that drake (and this repo) continue to move forward.  In particular, we've migrated to python3 completely now (and are almost finished moving the planar visualizer from the underactuated repo into the drake repo).  I think we will not dig in to try to uncover this sort of issue with the python2.7 version of the underactuated repo.

gizatt, 2019-11-14:
That said, this sounds like a memory leak within the container, so hopefully just restarting the container (and maybe your computer for good measure if that isn't enough) will get you back on track?

vinbo, 2019-11-15:
> 
> 
> That said, this sounds like a memory leak within the container, so hopefully just restarting the container (and maybe your computer for good measure if that isn't enough) will get you back on track?

yes, it recovers if I restart.

closed 2019-11-15 by RussTedrake