RussTedrake/underactuated #236: IOPub data rate exceeded

vinbo, 2019-11-17:
I am working on problem set 4, the plannar hopper.  When I run the test, it gives:

IOPub data rate exceeded.
The notebook server will temporarily stop sending output
to the client in order to avoid crashing it.
To change this limit, set the config variable
`--NotebookApp.iopub_data_rate_limit`.

Current values:
NotebookApp.iopub_data_rate_limit=1000000.0 (bytes/sec)
NotebookApp.rate_limit_window=3.0 (secs)

ps: my algorithm runs well if i pick up one initial condition from the test_set_4.py, like 0.5 m/s lateral velocity.

gizatt, 2019-11-18:
When I've seen this, it's always from too many error messages or print statements being dumped into the notebook by the code that's running -- so your idea that #237 is causing it seems pretty likely.

vinbo, 2019-11-19:
Yes, you are right. I printed too many log info. when I commented them, everything went normal. Thank you.

closed 2020-01-19 by RussTedrake