RussTedrake/underactuated #237: [IPKernelApp] WARNING | No such comm: eb8a242082d3404b8546d519c3a715b5

vinbo, 2019-11-17:
I'm working on pset4, hopper, every time I run my program in the sandbox, (not run test), the docker windows gives this info:

[2019-11-17 15:02:03.721] [console] [info] Adding joint body_x to the
[2019-11-17 15:02:03.721] [console] [info] Adding joint body_z to the
[2019-11-17 15:02:03.722] [console] [info] Adding joint body_theta to
[2019-11-17 15:02:03.723] [console] [info] Adding joint thigh to the p
[2019-11-17 15:02:03.724] [console] [info] Adding joint extension to t
[IPKernelApp] WARNING | No such comm: 3711d8e3fb504e63b708580401e055fe
[IPKernelApp] WARNING | No such comm: 8f500fecb54040e6acce879d05b4e38c
[IPKernelApp] WARNING | No such comm: d963468860044595b920a79558a4f119
[IPKernelApp] WARNING | No such comm: a2136af22a264e9d911809e6e436ee38
[IPKernelApp] WARNING | No such comm: 3711d8e3fb504e63b708580401e055fe
[IPKernelApp] WARNING | No such comm: 8f500fecb54040e6acce879d05b4e38c
[IPKernelApp] WARNING | No such comm: d963468860044595b920a79558a4f119
[IPKernelApp] WARNING | No such comm: a2136af22a264e9d911809e6e436ee38
[IPKernelApp] WARNING | No such comm: eb8a242082d3404b8546d519c3a715b5
[IPKernelApp] WARNING | No such comm: eb8a242082d3404b8546d519c3a715b5
[IPKernelApp] WARNING | No such comm: eb8a242082d3404b8546d519c3a715b5
[IPKernelApp] WARNING | No such comm: eb8a242082d3404b8546d519c3a715b5
[IPKernelApp] WARNING | No such comm: eb8a242082d3404b8546d519c3a715b5
[IPKernelApp] WARNING | No such comm: eb8a242082d3404b8546d519c3a715b5
[IPKernelApp] WARNING | No such comm: eb8a242082d3404b8546d519c3a715b5
[IPKernelApp] WARNING | No such comm: eb8a242082d3404b8546d519c3a715b5
[IPKernelApp] WARNING | No such comm: eb8a242082d3404b8546d519c3a715b5

this seems not affecting the result of my program. but i don't know if this relates to the issue below:
https://github.com/RussTedrake/underactuated/issues/236
so, ...

gizatt, 2019-11-18:
Is this new? (As in: has it always occured in Set 4 for you? Did it not occur in previous sets?)

vinbo, 2019-11-19:
It seems happens only in set4. I just tried pset3, it did not happen. (I'm in container drake-20190423).
And this happens every single time in set4.

TobiaMarcucci, 2020-03-22:
Closing this since we do not use the docker workflow anymore for the exercises.

closed 2020-03-22 by TobiaMarcucci