RussTedrake/underactuated #238: unexpected output of QP solver of MathematicalProgram

vinbo, 2019-11-23:
I think there is something wrong with the QP solver.
the example code in mathematical_program_and_sos_examples.ipynb, in pset3, the code segment

mp = MathematicalProgram()
x = mp.NewContinuousVariables(2, "x")
mp.AddConstraint((x**2).sum() == 1.)
mp.AddLinearCost(x.sum())
result = Solve(mp)
print result.is_success()
print result.GetSolution(x)
--------------------------------------
and
mp = MathematicalProgram()
x = mp.NewContinuousVariables(2, "x")
mp.AddConstraint(x.dot(x) == 1.)
mp.AddLinearCost(x.sum())
result = Solve(mp)
print result.is_success()
print result.GetSolution(x)
--------------------------------------------
They all give results: 
False
[ 0.  0.]
This is not right. So, what happened? Thanks

RussTedrake, 2019-11-23:
Can u please open this as an issue in drake instead?  There are many more people that can help there...

vinbo, 2019-11-23:
> 
> 
> Can u please open this as an issue in drake instead? There are many more people that can help there...

Do you mean here https://github.com/RobotLocomotion/drake ? I didn't notice this link before. And I even don't know whether their most updated master version still have the problem. Anyway, let's see what they will say.

vinbo, 2019-11-23:
> 
> 
> Can u please open this as an issue in drake instead? There are many more people that can help there...

https://github.com/RobotLocomotion/drake/issues/12397#issuecomment-557795702
well, the python3 verson does not have this problem...

vinbo, 2019-11-24:
> 
> 
> Can u please open this as an issue in drake instead? There are many more people that can help there...

Prof, someone over there really helps. By giving different initial guess, the solver's problem is clear:
1. result = Solve(mp)-----------gives:  False
2. result = result = Solve(mp,np.array([0.1,0.1])) ----------gives: 
True
[ 0.70710678  0.70710678]
3. other initial points not landing on x=y>=0 line give: 
True
[-0.70710678 -0.70710678]

So the solver itself is somewhat naive here. So I guess when this piece of code is used as example code in the jupyter notebook, a different solver was used?

RussTedrake, 2019-12-01:
I think the discussion on the drake issue were good/clear.  The solver is not wrong, the contract for a non-convex optimization is that the solver may incorrectly report infeasible.

This tutorial is going away, anyhow, since we have cleaned it up and ported a better version to drake master:
https://nbviewer.jupyter.org/github/RobotLocomotion/drake/blob/master/tutorials/mathematical_program.ipynb

closed 2019-12-01 by RussTedrake