RussTedrake/underactuated #240: what's the check of "system approximately obeys the original dynamics" in pset 4?

vinbo, 2019-11-28:
The test of problem 4 gave me this error:
Test Check that the system approximately obeys the original dynamics: 0.00/4.00.
  * Test Failed: index 50 is out of bounds for axis 0 with size 50

But I never changed the original dynamics of the system. And I cannot even find the words "out of bounds for axis" in the test_set_4.py. May I ask what this is for?

RussTedrake, 2019-12-02:
The issues you are opening are good, thank you.  But they reveal two deeper problems with our setup right now.
1) The psets are not kept up to date with drake, nor under CI.  We tend to update them each term, but not in between.
2) The psets aren't actually stored in this repository.   We have a private repo where we maintain them.  So when you add issues to this repo, mostly only I see them, but it's the TAs that have been writing those psets.

I think you reporting these now is still good.  The best solution, I think, is for me to move the problems we like most into the text and under CI, so they are kept better up to date.  We are also investigating what it would look like to have the docker instances maintained with the latest drake on CI.

gizatt, 2019-12-02:
BTW, I've already mentioned your reports here to the TA for the upcoming year, to be sure they're on the checklist to be resolved in the next class iteration.

You can open up `test_set_4.py` yourself and check out the details, but the gist is that this test confirms that the result of your trajectory optimization satisfies the dynamics constraint for each pair of knots. If I had to guess, you're getting an indexing bug when it tries to access `time_array[j+1]` for j=(the last sample point) -- double-check that your array of times is 1 longer than your array of states.

vinbo, 2019-12-03:
> BTW, I've already mentioned your reports here to the TA for the upcoming year, to be sure they're on the checklist to be resolved in the next class iteration.
>
> You can open up `test_set_4.py` [...]

Thanks. Actually, I kind of belive your guess about the index bound, by the literal meaning of the error info. but I checked each line of my code and printed the shapes, no one violated the index bound. Another proof is that I passed all the optimization problem in the test, with score 51/55. So this is kind of weird.
I already checked the test_set_4.py, there is no keywords involving "is out of bounds for axis" in there.

Anyway, this is not that important, maybe I should forget it.

TobiaMarcucci, 2020-03-22:
Closing this since the orbital transfer problem has been renewed here http://underactuated.mit.edu/trajopt.html#section8.

closed 2020-03-22 by TobiaMarcucci