RussTedrake/underactuated #241: About understanding the condition of grasp closure

vinbo, 2019-12-10:
In pset5, the givn conditions of grasp closure is :
*Check that 𝐺 is full row rank,
*find 𝑓 such that 𝐺𝑓=0
*and 𝑓 is in the interior of the set of allowable forces (i.e., in friction cone). Otherwise stated, 𝑓𝑖,𝑧>0, |𝑓𝑖,𝑥|<𝜇𝑓𝑖,𝑧  for all 𝑖.
I noticed that the 3rd condition is > and <, not the <=  as we model the friction using the coulomb model. 
I tried to changed the conditon3 into >= and <= in conditon 3, and found that in following case
[image]
(note the almost 0 friction, or even 0 friction)
the program recoginizes it as True, ie, grasp closure. But obviously it cannot resist the rolling torque of the cylinder. 
So the conclusion is, the original condition3 cannot be modified. But I just cannot understand why the condition is not the same as the Coulomb model. Will anyone help me understand this? Thanks

vinbo, 2019-12-21:
I think I know part of the answer now. The reason is that f here is inner force. It need to be >0 not >=0. but the rest has not been figured out

closed 2019-12-23 by vinbo

gizatt, 2019-12-24:
Sorry to not respond in a while -- I think you've got it. The forces need to have positive normal component, otherwise they're entirely zero (since you need normal force to apply tangential force) and useless to the grasp. In your case, I'm guessing you concluded your example returned a false positive with the modification because it was finding a set of zero forces?

The distinction about forces being in the _interior_ of the friction cone (i.e. |f_ix| < \mu f_iz) vs on the surface of the friction cone (|f_ix| = \mu f_iz) is subtle -- our convention is that forces in the interior are sticking, and forces on the surface of the friction cone are slipping. If you require that, then your case of having zero-volume friction cones (mu = 0) will never be solvable. In this particular example, I think it might be reasonable to relax that assumption and allow equality on the upper bound... but I'm not totally sure of that.

(Why do we have that convention? Well, if you want to reason about contact that is explicitly in a sliding mode for 
e.g. trajectory optimization, you want to constraint on the |f_ix| = \mu f_iz manifold. So then sticking is restricted to be the region inside of that.)