RussTedrake/underactuated #242: Why can the finger of gripper slip "into" the manipuland?

vinbo, 2019-12-21:
In the problem set 5, this situation sometimes happens: one of the fingers may suddenly slip into the ball, like this
[image]
Sometime it penetrates even deeper. I cannot understand why this happens. Any one has an answer? Thanks.

gizatt, 2019-12-21:
Judging by the arrows, your contact forces are very high. The penetration looks sort of like simulation steps being too high for the contact forces you're applying -- you could either decrease the sim step rate or find a way to limit the applied forces.

vinbo, 2019-12-22:
> 
> 
> Judging by the arrows, your contact forces are very high. The penetration looks sort of like simulation steps being too high for the contact forces you're applying -- you could either decrease the sim step rate or find a way to limit the applied forces.

Are you suggesting that the manipuland is not modeled as rigid body? Are you sure?

gizatt, 2020-01-03:
Rigid body collisions are modeled as stiff springs when the bodies are interpenetrating -- see [here](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#mbp_penalty_method), or the more complete treatment [here](https://drake.mit.edu/doxygen_cxx/group__drake__contacts.html). Any luck with decreasing the simulation timestep or otherwise regulating your contact forces?

RussTedrake, 2020-01-19:
Will close this due to inactivity.  Feel free to reopen if you think it is a real issue that warrants a resolution.

closed 2020-01-19 by RussTedrake