RussTedrake/underactuated #247: replace colab workflow with binder(?) to have drake already provisioned

RussTedrake, 2020-01-10:
I would ideally like to have "problems" at the end of each chapter that could open up a jupyter notebook and allow them to fill in some code and run their results.  students taking the class would then download the jupyter notebook and submit the notebook files to our autograder.

(we'd also like for folks to be able to do their psets locally, using the docker workflow, if they prefer -- or for more computationally demanding problems).

cc @TobiaMarcucci

RussTedrake, 2020-01-10:
fwiw -- the drake tutorials linked from the drake front page already use this workflow.

TobiaMarcucci, 2020-01-10:
I was looking at binder with @gizatt , and we noticed a couple of potential problems:

1. It seems that the only way students will have to save their work is to download a copy of the notebook they modified and then reupload it in the next session. From binder's docs, it seems that [after 10 minutes of inactivity all the modifications to a notebook are lost](https://mybinder.readthedocs.io/en/latest/faq.html#how-long-will-my-binder-session-last). This might be quite dangerous?
2. [The maximum number of simultaneous users for a given repo is 100](https://mybinder.readthedocs.io/en/latest/user-guidelines.html#maximum-concurrent-users-for-a-repository), which is pretty tight? It seems however that we could negotiate that number...

RussTedrake, 2020-01-11:
cc @EricCousineau-TRI

RussTedrake, 2020-01-14:
@jamiesnape -- curious to hear what you think?
(I will also check with our MIT CSAIL admins to see if we can host a JupyterHub)

jamiesnape, 2020-01-14:
Setting up a JupyterHub either at MIT or on AWS would probably work best for the class. If CSAIL cannot help, let me know and I can help with AWS.

There is still some utility in having Binder setup here (for casual users and non-MIT students following along), but it probably would not be optimal for MIT students submitting assignments.

RussTedrake, 2020-01-18:
I found this page helpful for jupyterhub vs colab: https://ithelp.brown.edu/kb/articles/jupyterhub-overview

TobiaMarcucci, 2020-01-18:
One consideration regarding JupyterHub. In order to control the access to the MIT servers, we will need to authorize the access of each single user (e.g., giving access for 4 months). This could work for the class, but probably wouldn't be great for people attending the class outside MIT? Every time that somebody wants to play with pydrake, he/she has to wait for our authorization. This might not be in line with the philosophy "make the access to the code as fast and simple as possible"?

On the Colab side, it seems that the timeouts are [90 minutes if you close the browser, 12 hours if you keep it open](https://stackoverflow.com/questions/54057011/google-colab-session-timeout). On average it take a couple of minutes to set up pydrake. Moreover, there don't seem to be strict constraints on computation power. Do you think this could be the best option for the class?

P.S.: One other issue with binder is that [once one exceeds 2GB of ram usage they automatically restart your kernel](https://mybinder.readthedocs.io/en/latest/faq.html#how-much-memory-am-i-given-when-using-binder).

jamiesnape, 2020-01-20:
> One consideration regarding JupyterHub. In order to control the access to the MIT servers, we will need to authorize the access of each single user (e.g., giving access for 4 months). This could [...]

I think we can run a version on Binder regardless. It is simple to do and low overhead.

TobiaMarcucci, 2020-01-20:
I agree: I also think that for tutorials, where there isn't much to code, Binder is cleaner than Colab or similars.

RussTedrake, 2020-01-20:
as I just realized

EricCousineau-TRI, 2020-01-21:
> I think we can run a version on Binder regardless. It is simple to do and low overhead.

Just to clarify: Do you mean to keep the current Drake tutorial setup for Binder, or do you mean hosting a JupyterHub instance for use with the class, but possibly using resources on AWS vs. MIT servers?

jamiesnape, 2020-01-21:
Binder for this repo. Drake would not change.

EricCousineau-TRI, 2020-01-21:
Gotcha. And do you mean using public / free Binder hosting, or standing up a separate JupyterHub instance?

jamiesnape, 2020-01-21:
It might be both. A osted one for MIT and a public/free one for non-MIT. Depends on CSAIL, but, regardless, I would set up a public one as it is so easy.

commit bb51749e (2020-02-02): Update appendix to reflect the jupyter-first workflow, and give some info about colab vs binder.
Resolves #247.
PR #271 (2020-02-02): Update appendix to reflect the jupyter-first workflow,  …
and give some info about colab vs binder.

Resolves #247.

---
This change is [[image]](https://reviewable.io/reviews/russtedrake/underactuated/271)
commit c0b7f270 (2020-02-02): Update appendix to reflect the jupyter-first workflow, and give some info about colab vs binder. (#271)

Resolves #247.
closed 2020-02-02 by RussTedrake