RussTedrake/underactuated #248: course docker workflow should use the drake docker images.

RussTedrake, 2020-01-10:
as discussed with cc @jamiesnape 
will assign to @tobiamarcucci

TobiaMarcucci, 2020-01-22:
A clarification:

Do you mean that the Docker image currently in the [`6-832-backend` repo](https://github.com/RobotLocomotion/6-832-backend/blob/wei_docker_test/docker/Dockerfile.ubuntu16.04) should inherit from [Drake's Docker image](https://github.com/RobotLocomotion/drake/blob/master/setup/ubuntu/docker/bionic/Dockerfile)? If yes, that could be tricky since for the autograding we need the Docker file to inherit from Gradescope. [This section of the Docker docs](https://docs.docker.com/develop/develop-images/multistage-build/) might be relevant.

Also, do we plan the  `underactuated` repo to have its own Docker file?

jamiesnape, 2020-01-22:
> Do you mean that the Docker image currently in the 6-832-backend repo should inherit from Drake's Docker image? If yes, that could be tricky since for the autograding we need the Docker file to inherit from Gradescope. This section of the Docker docs might be relevant.

Whatever docker image the students use to follow along with the class and complete their 
assignments.

> Also, do we plan the underactuated repo to have its own Docker file?

Probably, unless there is good reason not to. Binder would need some sort of Dockerfile, regardless.

RussTedrake, 2020-02-06:
Updates:
- Binder is up -- using only the drake docker + a local clone of underactuated.
- Gradescope docker we understand must remain different, but only the staff should ever have to worry about that image.
- We are hoping to get well into the term without needing to distribute any docker; I suspect even as projects come around we can probably simply point people to use the drake docker (and clone underactuated only if they need the one or two tools that it still provides)

I think it's safe to close this for now.

closed 2020-02-06 by RussTedrake