RussTedrake/underactuated #258: Update / simplify user install instructions

RussTedrake, 2020-01-24:
Once we finish the CI/bazel churn, we need to make sure we're still happy with:
http://underactuated.csail.mit.edu/underactuated.html?chapter=drake

And perhaps we have a chance to simplify it?  

For instance, I was going to teach bazel to fetch drake itself following https://github.com/RobotLocomotion/drake-external-examples/blob/master/drake_bazel_external/WORKSPACE#L57 , but realize now that the setup process needs to download drake to run install_prereqs, first.

jamiesnape, 2020-01-24:
I can teach Bazel to fetch Drake (`drake-external-examples` is incomplete), but it might be easier to download it, since then the workflow could just use the (forthcoming) Docker images with drake/underactuated preinstalled.

RussTedrake, 2020-01-24:
Good.  You’re right that I’m carrying too many workflows.  The docker one is separate still.

I agree - having the docker workflow (based on drake docker, not gradescope docker) be the main recommended workflow for the course on the page linked above sounds right.  I can include links to the drake install docs for people who want to explore anything else (+ a note that we have install prereqs in this repo, too).

RussTedrake, 2020-04-04:
the appendix is simpler than it was, and I don't feel an immediate action item here.  closing this for now.

closed 2020-04-04 by RussTedrake