RussTedrake/underactuated #273: Binder provisioning wastes a LOT of time downloading mathjax

RussTedrake, 2020-02-02:
There is no reason for the binder docker image to have to checkout the htmlbook submodule.  If we can disable that, Binder will load a lot faster.  (Hopefully this is easy?)

I would say Binder/Colab performance is my top priority as it effects the "first impression" of users, and classes start on Tuesday.

jamiesnape, 2020-02-03:
Looks like https://github.com/jupyter/repo2docker/pull/233 is the main problem.

RussTedrake, 2020-02-05:
Plan:
- [ ] zap the mathjax submodule from htmlbook
- [ ] point underactuated.html to either a cdn or the mathjax repo hosted by itself on underactuated.
- [ ] to support offline writing, i can implement a [fallback to a local checkout](https://github.com/mathjax/MathJax-docs/wiki/How-to:-MathJax-CDN-fallback) (presumably htmlbook is still a good place for that, but I can just add a `Readme.md` that tells people to check it out manually).

commit 0bc07a9e (2020-02-06): Remove local mathjax

Resolves #273
(note that I am depending on my server having still checked cloned things locally)
closed 2020-02-06 by RussTedrake