RussTedrake/underactuated #297: binder is broken?

RussTedrake, 2020-02-11:
Loading my very [first example](https://mybinder.org/v2/gh/RussTedrake/underactuated/master?filepath=examples%2Fdouble_pendulum%2Fdynamics.ipynb) results in:

```
Waiting for build to start...
Picked Git content provider.
Cloning into '/tmp/repo2dockersek_05t2'...
HEAD is now at dca64fb binder supports notebook backend
Submodule 'htmlbook' (http://github.com/RussTedrake/htmlbook) registered for path 'htmlbook'
Cloning into '/tmp/repo2dockersek_05t2/htmlbook'...
warning: redirecting to https://github.com/RussTedrake/htmlbook/
Submodule path 'htmlbook': checked out '4727f149b9a3465fe34a29038c5db991024c1199'
Using DockerBuildPack builder
Step 1/15 : FROM robotlocomotion/drake:latest
 ---> 789902819e0fe...
Step 2/15 : ARG NB_USER=jovyan
 ---> Running in a23ea021c1f1
Removing intermediate container a23ea021c1f1
 ---> 710e6de0e76c
Step 3/15 : ARG NB_UID=1000
 ---> Running in 7203172b73e7
Removing intermediate container 7203172b73e7
 ---> 6e4bd5ee6a99
Step 4/15 : ARG NB_GID=100
 ---> Running in ca7e61405d85
Removing intermediate container ca7e61405d85
 ---> b6fde60add6e
Step 5/15 : EXPOSE 8888/tcp
 ---> Running in 2ea72cf774b5
Removing intermediate container 2ea72cf774b5
 ---> 6b4bfd0a25ae
Step 6/15 : COPY scripts/setup/ubuntu/18.04/install_prereqs /tmp/
 ---> 8090a6de8b29
Step 7/15 : RUN export DEBIAN_FRONTEND=noninteractive   && /tmp/install_prereqs > /dev/null 2>&1   && rm -rf /tmp/install_prereqs /var/lib/apt/lists/*   && useradd -d /home/$NB_USER -G $NB_GID -mU -s /bin/bash $NB_USER
 ---> Running in bc771f3acd06
Failed to connect to event stream
```

jamiesnape, 2020-02-11:
`Failed to connect to event stream` is a client-side problem. Everything is fine for me. Try clearing your browser cache or try a different browser (for Binder, Firefox seems most reliable, followed by Chrome, followed by Safari).

RussTedrake, 2020-02-11:
cc @TobiaMarcucci who has been having similar issues.

RussTedrake, 2020-02-11:
(Mine did work again after a few refreshes)

TobiaMarcucci, 2020-02-11:
Now it's working for me too. But this morning (and many times in the last week) I had the same `Failed to connect to the event stream` error, but I didn't even have the steps that Russ reported. It was a single line build log with that message.

RussTedrake, 2020-02-13:
Let's use this issue to continue to monitor the situation.

lyltc1, 2020-03-18:
I'm sorry that I have problem using Binder. What should I do to solve it?

> Waiting for build to start...
Picked Git content provider.
Cloning into '/tmp/repo2dockerkanyy8ia'...
HEAD is now at 7584a59 Rephrases Exercise 7.1e (#334)
Submodule 'htmlbook' (http://github.com/RussTedrake/htmlbook) registered for path 'htmlbook'
Cloning into '/tmp/repo2dockerkanyy8ia/htmlbook'...
warning: redirecting to https://github.com/RussTedrake/htmlbook/
Submodule path 'htmlbook': checked out 'e9b87a66de3f4688d73e14fc190bc4b5da54b2f7'
Using DockerBuildPack builder
Step 1/15 : FROM robotlocomotion/underactuated:latest
pull access denied for robotlocomotion/underactuated, repository does not exist or may require 'docker login': denied: requested access to the resource is denied

jamiesnape, 2020-03-18:
Try again now; there was an issue with the base image, which has now been refreshed.

closed 2021-04-13 by RussTedrake