RussTedrake/underactuated #348: install_prereqs bazel version is fighting drake's install_prereqs

RussTedrake, 2020-04-02:
After running the install script in underactuated, i had an updated bazel, which was then downgraded by running install_prereqs again in drake.  

@jamiesnape -- might you recommend the best practices solution?

jamiesnape, 2020-04-03:
Drake is fighting apt, so the best practice solution would be for Drake to not do that. However, that may not be palatable to Drake (though I think Bazel may be mature enough for that to be mostly fine now). Therefore, several other options:

1. Add Bazel to the `install_prereqs` that is bundled with the Drake binaries (we add CMake and compilers already, for instance).
2. Bundle another script with Drake that installs the correct development tools (leave the libraries, Python modules, etc. in the current `install_prereqs` (has other, unrelated, benefits).
3. Use `dpkg` instead of `apt` in underactuated and try to keep in sync with the code in Drake.
4. Answer "no" when Drake asks to downgrade Bazel.

RussTedrake, 2020-04-05:
for 4 -- i don't actually get asked.  the install_prereqs script runs without user input, i think?
3 might be the path of least resistance for now?  we could presumably do this only for bazel?

jamiesnape, 2020-04-06:
Must be an issue with https://github.com/RobotLocomotion/drake/blob/b92b1f678aad850c2b55ce36af8fa17d02045876/setup/ubuntu/source_distribution/install_prereqs.sh#L47-L56

closed 2022-02-02 by RussTedrake