RussTedrake/underactuated #353: widgets are not working on binder

RussTedrake, 2020-04-04:
https://mybinder.org/v2/gh/RussTedrake/underactuated/master?filepath=examples%2Fautapse_and_lstm.ipynb

running the first and second cells shows the error.  
```
Widget Javascript not detected.  It may not be installed or enabled properly.
```

the browser javascript console shows errors about the python versions.  
```
main.js:230 Widgets are not available.  Please install widgetsnbextension or ipywidgets 4.0
```

Presumably this broke when we switched to the underactuated dockerfile, but i didn't notice until now.

RussTedrake, 2020-04-04:
Adding a cell with
```
import ipywidgets; print(ipywidgets.__version__)
import widgetsnbextension; print(widgetsnbextension.__version__)
```
reveals 
```
6.0.0
2.0.0
```
which is not what we have in `requirements.txt`

jamiesnape, 2020-04-06:
But it is what `apt` installs, so that is correct. Why it is not working is something I need to look into.

jamiesnape, 2020-04-06:
Looks like some issue with `jupyter nbextension enable --py widgetsnbextension`.

RussTedrake, 2020-04-06:
thanks for working on it.

commit 154190c6 (2020-06-06): Ensure widgetsnbextension is enabled for all users

Closes #353.
closed 2020-06-06 by jamiesnape