RussTedrake/underactuated #358: Make the auto grading look nicer

TobiaMarcucci, 2020-04-12:
@wilkosch: I think it'd be nice to make the autograding cell at the end of the notebooks look nicer. What do you think?

For example, would the addition of some `__init__.py` files prevent us from writing the long import lines? Also, can the `Grader` class provide a single function that wraps the actual grading and the printing?

One more related thing. Is there a reason why you decided not to delete the `results.json` file after you run the grader? Every time I run the grader, I get an unstaged file in git. We could add `.json` to the `.gitignore`, but I think it'd be cleaner to just delete the `json`s after grading.

wilkosch, 2020-04-12:
Sure, I agree with you!

Overall my thoughts were that readability > less code / fewer lines.
I wanted to explicitly show students where the tests are, so they can debug it themselves when certain tests are failing. That's why I wanted to avoid stashing imports into init files. We can easily change it though. What do you think?

I agree with you that the last two lines could be turned into one and delete the `results.json` automatically. My thinking was that I wanted to allow students to be able to upload them to us if things go wrong.  Nevertheless, I don't think we have really used this method of debugging since gradescope already shows us the output.

TobiaMarcucci, 2020-04-12:
I see. For the second point, they've always sent us screenshots or s copy and paste of the error message. So I think it's safe to just delete the json.

For the first I agree that telling them where the tests are is good for debugging. But Couldn't we do it explicitly? Like adding a sentence to the markdown above the final cell saying something like "You can check your work by running the following cell. This will grade your implementation running the battery of tests written in `this_file`.", with `this_file` being a link that points to the test file in GitHub?

RussTedrake, 2025-01-03:
I'm going to close this until we find someone that wants to fix it.

closed 2025-01-03 by RussTedrake