RussTedrake/underactuated #367: Double Pendulum ImportError libClp.so.1

the-raspberry-pi-guy, 2020-04-25:
Hi there,

I am really enjoying going through the course so far and I am just trying to get the course Jupyter Notebook resources to run on my Ubuntu 18 computer. Following the instructions at: http://underactuated.mit.edu/drake.html means that I can import pydrake with no issues, however when I try to run the Juypter Notebook double pendulum example I come across an ImportError with libClp.so.1:

[image]

I am not sure what this is? Is there potentially an issue with my configuration? Any help is appreciated and thanks for the course :)

RussTedrake, 2020-04-25:
libClp comes from the drake COIN dependency, I believe.  If you've run the `install_prereqs` script and it passed successfully, then this is an issue that we should understand/resolve in drake.  

Could you [post an issue on stackoverflow](https://stackoverflow.com/questions/ask?tags=drake) with the drake tag?

the-raspberry-pi-guy, 2020-04-25:
My apologies, I was clearly too hasty to post this issue. While the install_prereqs script for textbook supplement executed correctly, the install_prereqs for Drake missed a few dependencies due to HDD capacity on my machine.

Starting the installation process again from scratch led to no errors:
[image]

Thanks for the quick reply!

closed 2020-04-25 by the-raspberry-pi-guy