RussTedrake/underactuated #372: drake's gurobi license acquisition does not work via bazel test

RussTedrake, 2020-05-24:
```
russt@Puget-179850-01:~/underactuated$ bazel test //...
INFO: Invocation ID: 782d9316-41dc-48e3-bb86-5f34c91386e6
INFO: Analyzed 330 targets (0 packages loaded, 0 targets configured).
INFO: Found 99 targets and 231 test targets...
FAIL: //underactuated:mathematical_program_examples_ipynb (see /home/russt/.cache/bazel/_bazel_russt/6a53b066a912ced1e037beb4a17598ee/execroot/underactuated/bazel-out/k8-fastbuild/testlogs/underactuated/mathematical_program_examples_ipynb/test.log)
INFO: From Testing //underactuated:mathematical_program_examples_ipynb:
==================== Test output for //underactuated:mathematical_program_examples_ipynb:
Traceback (most recent call last):
  File "/home/russt/.cache/bazel/_bazel_russt/6a53b066a912ced1e037beb4a17598ee/sandbox/linux-sandbox/514/execroot/underactuated/bazel-out/k8-fastbuild/bin/underactuated/mathematical_program_examples_ipynb.runfiles/underactuated/underactuated/mathematical_program_examples_ipynb.ipynb.py", line 109, in <module>
    result = Solve(mp)
RuntimeError: Could not create Gurobi environment because Gurobi returned code 10009 with message "HostID mismatch (licensed to bfcd5718, hostid is 0)".
================================================================================
FAIL: //examples:trajectory_optimization (see /home/russt/.cache/bazel/_bazel_russt/6a53b066a912ced1e037beb4a17598ee/execroot/underactuated/bazel-out/k8-fastbuild/testlogs/examples/trajectory_optimization/test.log)
INFO: From Testing //examples:trajectory_optimization:
==================== Test output for //examples:trajectory_optimization:
Traceback (most recent call last):
  File "/home/russt/.cache/bazel/_bazel_russt/6a53b066a912ced1e037beb4a17598ee/sandbox/linux-sandbox/515/execroot/underactuated/bazel-out/k8-fastbuild/bin/examples/trajectory_optimization.runfiles/underactuated/examples/trajectory_optimization.ipynb.py", line 61, in <module>
    result = Solve(prog)
RuntimeError: Could not create Gurobi environment because Gurobi returned code 10009 with message "HostID mismatch (licensed to bfcd5718, hostid is 0)".
================================================================================
FAIL: //examples:miqp_planning (see /home/russt/.cache/bazel/_bazel_russt/6a53b066a912ced1e037beb4a17598ee/execroot/underactuated/bazel-out/k8-fastbuild/testlogs/examples/miqp_planning/test.log)
INFO: From Testing //examples:miqp_planning:
==================== Test output for //examples:miqp_planning:
Traceback (most recent call last):
  File "/home/russt/.cache/bazel/_bazel_russt/6a53b066a912ced1e037beb4a17598ee/sandbox/linux-sandbox/513/execroot/underactuated/bazel-out/k8-fastbuild/bin/examples/miqp_planning.runfiles/underactuated/examples/miqp_planning.ipynb.py", line 54, in <module>
    result = Solve(prog)
RuntimeError: Could not create Gurobi environment because Gurobi returned code 10009 with message "HostID mismatch (licensed to bfcd5718, hostid is 0)".
================================================================================
INFO: Elapsed time: 3.340s, Critical Path: 2.42s
INFO: 6 processes: 1 remote cache hit, 5 linux-sandbox.
INFO: Build completed, 3 tests FAILED, 4 total actions
//examples:miqp_planning                                                 FAILED in 2.1s
  /home/russt/.cache/bazel/_bazel_russt/6a53b066a912ced1e037beb4a17598ee/execroot/underactuated/bazel-out/k8-fastbuild/testlogs/examples/miqp_planning/test.log
//examples:trajectory_optimization                                       FAILED in 1.7s
  /home/russt/.cache/bazel/_bazel_russt/6a53b066a912ced1e037beb4a17598ee/execroot/underactuated/bazel-out/k8-fastbuild/testlogs/examples/trajectory_optimization/test.log
//underactuated:mathematical_program_examples_ipynb                      FAILED in 0.6s
  /home/russt/.cache/bazel/_bazel_russt/6a53b066a912ced1e037beb4a17598ee/execroot/underactuated/bazel-out/k8-fastbuild/testlogs/underactuated/mathematical_program_examples_ipynb/test.log
```

Two things changed in drake since I tried it before: @jwnimmer-tri updated the license mechanism, and @hongkai-dai updated us to gurobi 9.0.2.  

I verfied that i can successfully run the code in the jupyter notebook, and successfully run the same code from a python script (not launched via bazel).  My best guess is that whatever mechanism (environment variable?) that gurobi is using to get the host id is not being passed through to the bazel test in here: https://github.com/RussTedrake/underactuated/blob/master/.bazelrc#L2

cc @toffeealbina-tri

RussTedrake, 2020-05-24:
> hostid 0 means your computer has no network adaptor.  https://groups.google.com/forum/#!topic/gurobi/tllLMdiVqhc

RussTedrake, 2020-05-24:
Just found this in `drake/tools/skylark/test_tags.bzl`
```
Gurobi checks a license file outside the workspace so tests that use Gurobi
    must have the tag "no-sandbox". For the moment, we also require the tag
    "exclusive" to rate-limit license servers with a small number of licenses.
```
manually adding 
```
    tags = ["no-sandbox"],  # For gurobi license.
```
to the failing tests seems to resolve the problem.  (but yuck!)

closed 2020-05-24 by RussTedrake

jamiesnape, 2020-05-26:
`--spawn_strategy=standalone` when using Gurobi might be slightly easier?

jwnimmer-tri, 2020-05-26:
FWIW In Anzu, I'm prototyping some new Gurobi integration.  In that work, the license file is part of the repository rule / build system integration directly.  Depending on what I learn there, we could try bringing some of it back into Drake.

jamiesnape, 2020-05-26:
I forgot before, but we have `block-network` by default here since Gurobi support was not something that seems relevant with only using binaries. Not sure if that makes a difference with respect to getting the hostid. If adding the tag `requires-network` fixes the problem, then we could change the default.

RussTedrake, 2020-05-26:
Changing "no-sandbox" out for "requires-network" also solved the problem.  That seems like a significantly better solution... no?  Should we fix that in drake?

RussTedrake, 2020-05-26:
fwiw -- i tried changing `no-sandbox` to `requires-network` in `gurobi_test_tags` in drake, and all of my tests pass.  but the explanation given is because it is accessing a license file outside of the workspace, which is still valid... so I will leave it to @jamiesnape and/or @jwnimmer-tri to decide and resolve.

jamiesnape, 2020-05-27:
Probably the combination of Jeremy effectively adding the license to the workspace and allowing network connections (which is probably the default in Drake) instead of disabling the sandbox is probably going to be the way forward.

You can always allow network connections by modifying here: https://github.com/RussTedrake/underactuated/blob/c714e945cb9cd940d96ebee3dee24c806a2866a4/tools/rt/python/defs.bzl#L19-L24

Similar logic can be applied to always switch off the sandbox for the python and jupyter tests.

I could probably give you a clean way to enable Gurobi and adjust the sandbox with a single command line switch, but I have quite a backlog at the moment.