RussTedrake/underactuated #386: Underactuated Setup for Ubuntu 20.04 Focal

MAminSFV, 2020-08-24:
It seems like the underactuated course supplement does not include an "install_prereqs.sh file" for ubuntu 20.04.

I was following the instructions on this page to run the  course material locally:

http://underactuated.csail.mit.edu/drake.html

Is it possible to use the course supplement on ubuntu focal in an alternative way?

Note: I have completed the binary setup for pydrake based on:
https://drake.mit.edu/python_bindings.html

2shaar2059, 2020-09-04:
As I was trying to build drake, I was getting dependency errors, specifically the libmumps one. Anyone else getting this?

KeranYe, 2021-07-31:
@MAminSFV Did you resolve the problem? If so, could you share the solution?

RussTedrake, 2021-07-31:
If you have problems installing drake, please post on stackoverflow. The
developers are quick to answer.

RussTedrake, 2022-02-02:
20.04 is fully supported.

closed 2022-02-02 by RussTedrake