RussTedrake/underactuated #393: Maybe there are errors in Chapter3.

zapplelove, 2020-12-02:
I re-calculate the equations of motion of the acrobot and I have a different answer to this and I don't know if I am wrong. Here is My code :
```matlab
syms l_c1 l_c2 l1 t1(t) t2(t)
P_c1=l_c1*[sin(t1);-cos(t1)];
P_c2=l1*[sin(t1);-cos(t1)]+l_c2*[sin(t1+t2);-cos(t1+t2)];

v_c1=diff(P_c1);
v_c2=diff(P_c2);
w1=diff(t1);
w2=diff(t2);

syms m1 m2 i1 i2
T1=1/2*m1*(v_c1.')*v_c1+1/2*(w1.')*i1*w1;
T2=1/2*m2*(v_c2.')*v_c2+1/2*(w2.')*i2*w2;
T=simplify(T1+T2);
syms dq1 dq2 q1 q2
T_sub=subs(subs(T,[w1,w2],[dq1,dq2]),[t1,t2],[q1,q2]);
I1=i1+m1*l_c1^2;
I2=i2+m2*l_c2^2;

T_r=1/2*I1*dq1^2 ...
    +1/2*(m2*l1^2+I2+2*m2*l1*l_c2*cos(q2))*dq1^2 ...
    +1/2*I2*dq2^2 ...
    +(I2+m2*l1*l_c2*cos(q2))*dq1*dq2;
simplify(T_r-T_sub)
```

The difference between the `T_r` and `T_sub` is
[image]

Above section 3.6, The actuator command is
[image]
This is different from the eq. (35), maybe this is
[image]

zapplelove, 2020-12-03:
> ```matlab
> w2=diff(t2);
> ```
I find out the error, this is 
 ```matlab
 w2=diff(t1)+diff(t2);
 ```

RussTedrake, 2022-02-02:
That's correct.  You had written the torque on the absolute angle, instead of the relative angle.  I see that posted the second possible error in hypothesis, too.  I will look/respond there.

closed 2022-02-02 by RussTedrake