RussTedrake/underactuated #401: purge scipy dependency

RussTedrake, 2021-02-18:
there is only one use case I can see, and I can't believe it's necessary.

RussTedrake, 2021-02-18:
@TobiaMarcucci -- The usage in https://github.com/RussTedrake/underactuated/blob/master/exercises/trajopt/shooting_vs_transcription/shooting_vs_transcription.ipynb is the only one I can see.  It looks like scipy is the typical recommendation for sparse linear algebra solves... but might you take one more look and see if it's necessary?  (it's a relatively heavy requirement for this one usage)

TobiaMarcucci, 2021-02-19:
I agree that if `scipy` is needed only there it's not worth. I had a look online and there are a few libraries for sparse linear algebra in python: it seems that the best alternative would be [pysparse](http://pysparse.sourceforge.net)? It doesn't seem too heavy of a dependency.

On the other hand I'm not sure if by using dense matrices that problem would be infeasible (i.e. use `numpy` arrays directly). At the time I chose sparse algebra bcs I thought it was nice for the students to be exposed to it and also because `scipy` provides a super nice interface for block matrices which made the code extremely clean: see [this construction](https://github.com/RussTedrake/underactuated/blame/master/exercises/trajopt/shooting_vs_transcription/shooting_vs_transcription.ipynb#L623).

I see two alternatives:

- Try to use normal `numpy` arrays and see if it runs in decent time. If yes, some sugar might be needed to provide to the students a simpler way to construct block matrices than the one of `numpy`... Otherwise I think most of them will just get stuck with the multiple `np.hstack`, `np.vstack`, and the sizes of the zero matrices. Maybe then mention in the pset that this can be done more efficiently using sparse algebra.
- If the first doesn't work, one can try `pysparse` above, but I'm not sure how well it works. Also here some sugar will be needed, it seems `pysparse` doesn't handle block matrices nicely: it seems [normal indexing is the suggested way](http://pysparse.sourceforge.net/Pysparse.pdf)...

RussTedrake, 2021-02-19:
Thanks Tobia.  I would vote against adding pysparse if it's not beautiful.  I would rather carry scipy forward given it might have other uses than add a new/different dependency.

I just looked to see if drake provides bindings for any of the eigen sparse matrix tools.  And see now that drake packages scipy.  I guess it is no problem/burden then for this repo to have scipy.  Mea culpa!

closed 2021-02-19 by RussTedrake