RussTedrake/underactuated #414: Notebook Setup failed on macOS 11.1

huangya17, 2021-02-24:
Hello, 

Could you point me to the right direction as how to run the example on broswers / Notebook? 

I have following the instruction on 
https://colab.research.google.com/github/RussTedrake/underactuated/blob/master/intro.ipynb#scrollTo=A4QOaw_zYLfI

But the installation stage failed persistently. Please see attached screen shot.

[image]

Thank you

Ya

RussTedrake, 2021-02-24:
You're doing the right thing.  Google colab upgraded from python3.6 to python3.7 yesterday and broke everything.  We are working feverishly to repair it.

huangya17, 2021-02-24:
Thank you for the lightning reply! 

I will keep up my pace with the notes and video lectures.

Ya

RussTedrake, 2021-02-27:
This should be completely resolved now.  (but please let me know if you find a notebook that fails).

closed 2021-02-27 by RussTedrake

huangya17, 2021-02-27:
Many thanks! I have tried codes on ch1 ch2 and ch7 with no problem.
Bests,
Ya