RussTedrake/underactuated #440: LittleDog.urdf textures do not load in meshcat

RussTedrake, 2021-04-18:
To reproduce, run the first example (standing with PID control).  in the notebook notebook `examples/littledog.ipynb`.  The texture maps are dark, but not that dark.  The eyes should be red.

cc @gizatt -- in case you have time to look.

gizatt, 2021-04-18:
Haven't tried running the code yet, but I'm suspect this is gated behind [meshcat-python's relatively weak texture support](https://github.com/rdeits/meshcat-python/issues/27). IIRC, MeshcatVisualizer hacks around this by looking for a `.png` texture file next to each `.obj` file and loading it in manually (blindly assuming the `.obj` specifies the UV stuff correctly). As Robin mentioned in that issue, `meshcat-julia` does texture loading automagically, so it's totally possible, but `meshcat-python` needs a substantial update. I've been meaning to tackle that for a long time, but haven't had an excuse yet...

Beyond that, the littledog resources may not be complete? (Even if they were, it wouldn't work because of the above.) E.g. [this file](https://github.com/RussTedrake/underactuated/blob/master/underactuated/models/littledog/meshes/back_left_lower.obj.mtl) references some jpgs that I don't see in the repo.

If the default color is really bad, you might be able to hack around for the moment my setting material properties directly in the urdf?

RussTedrake, 2021-04-19:
Thanks for taking a look.  I've located and adding the missing .jpg files, and verified that adding this line 
```
    dvis = DrakeVisualizer.AddToBuilder(builder, scene_graph=scene_graph)
```
just below the call to `ConnectMeshcatVisualizer` (+ the import for `DrakeVisualizer`) does allow me to view things with the expected textures in `drake_visualizer`.

RussTedrake, 2022-02-02:
i have this working in the new version of meshcat

closed 2022-02-02 by RussTedrake