RussTedrake/underactuated #453: Error installing Drake in google colab

ParthaShuvro, 2021-12-24:
I am not entirely sure about the issue but below is the detail that comes up once I try to install drake in colab for last few weeks. 

ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
torchvision 0.11.1+cu111 requires torch==1.10.0, but you have torch 1.8.1 which is incompatible.
torchtext 0.11.0 requires torch==1.10.0, but you have torch 1.8.1 which is incompatible.
torchaudio 0.10.0+cu111 requires torch==1.10.0, but you have torch 1.8.1 which is incompatible.
albumentations 0.1.12 requires imgaug<0.2.7,>=0.2.5, but you have imgaug 0.2.9 which is incompatible.

**E: Failed to fetch http://security.ubuntu.com/ubuntu/pool/universe/x/xorg-server/xvfb_1.19.6-1ubuntu4.9_amd64.deb  404  Not Found [IP: 91.189.88.152 **80]****
**E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?**

---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-1-d69cf667c2ae> in <module>()
      8                 "setup_underactuated_colab.py")
      9     from setup_underactuated_colab import setup_underactuated
---> 10     setup_underactuated(underactuated_sha='e089af19ca62590e131ad295b51d2c950a5ecc10', drake_version='0.27.0', drake_build='release')
     11 
     12 server_args = []

1 frames
/content/setup_underactuated_colab.py in run(cmd, **kwargs)
     94         if cp.stderr:
     95             print(cp.stderr)
---> 96         assert cp.returncode == 0, cp
     97 
     98     # Clone the repo (if necessary).

AssertionError: CompletedProcess(args=['apt', 'install', 'xvfb'], returncode=100, stdout='Reading package lists...\nBuilding dependency tree...\nReading state information...\nThe following NEW packages will be installed:\n  xvfb\n0 upgraded, 1 newly installed, 0 to remove and 37 not upgraded.\nNeed to get 784 kB of archives.\nAfter this operation, 2,270 kB of additional disk space will be used.\nIgn:1 http://archive.ubuntu.com/ubuntu bionic-updates/universe amd64 xvfb amd64 2:1.19.6-1ubuntu4.9\nErr:1 http://security.ubuntu.com/ubuntu bionic-updates/universe amd64 xvfb amd64 2:1.19.6-1ubuntu4.9\n  404  Not Found [IP: 91.189.88.152 80]\n', stderr='\nWARNING: apt does not have a stable CLI interface. Use with caution in scripts.\n\nE: Failed to fetch http://security.ubuntu.com/ubuntu/pool/universe/x/xorg-server/xvfb_1.19.6-1ubuntu4.9_amd64.deb  404  Not Found [IP: 91.189.88.152 80]\nE: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?\n')

cporoske, 2021-12-24:
Today I have the same problem as you.
I noticed that the error prompted **apt-get update**, so I created a new cell at the top of ` colab` and typed
**!sudo apt-get update**
Then continue with the following steps, the operation was successful. 
Wish to help u.

ParthaShuvro, 2021-12-24:
Thanks, saved the day.

closed 2021-12-24 by ParthaShuvro

DanielTakeshi, 2021-12-27:
Also worked for me thanks @cporoske  👍