RussTedrake/underactuated #454: Accessing ngrok links from colab

ParthaShuvro, 2022-01-02:
Can anyone please confirm if the ngrok links that are being generated in colab examples working? I cannot access them although I followed the instructions as provided in the ngrok documentation to authenticate and the local client app seems to work fine. If yes, please let me know if I am missing something.

RussTedrake, 2022-01-03:
I just confirmed that I also see an authentication page instead of the proper ngrok url.  Last time this happened, it was a mistake by the ngrok team; they had accidentally disabled all free accounts.  I will look into it more soon.

Last semester, I ported my manipulation class from colab to deepnote in part so that I could avoid this ngrok dependency.  I plan to start porting the underactuated notebooks to deepnote very soon (in time for this spring semester).

russorat, 2022-01-03:
Hi, I'm a product manager for ngrok. In an effort to curb abuse, we have made a change where users are required to register before serving html content via ngrok. Serving other content should not be impacted.

Is this the change you are talking about? or are you seeing something else? If you post a screenshot, it might help clear things up.

RussTedrake, 2022-01-03:
Hi @russorat.  Thanks for responding here!  Yes, this change from unregistered usage to requiring a registration is what I was talking about.  It's a major change to the workflow in these notebooks, which have so far been easy for anyone to open a browser and start running the course software.  Now they have to stop and register on yet another site to get going.

Moreover, since ngrok's [default behavior for http is to open two tunnels, and the free account is limited to one tunnel](https://pyngrok.readthedocs.io/en/latest/#open-a-tunnel), I'll need to update code much further upstream (the `meshcat-python` backend; I've already made this change in my c++ meshcat).  That's painful because it will have to filter through a month's worth of pip releases before it becomes properly available.  @russorat -- this seems like something that everyone using the free account will trip over?

@ParthaShuvro -- I think the new header for colab will have to look something like
```
!pip install underactuated==2022.01.03
from pyngrok import ngrok

# ngrok now requires users to register for a free account:
# https://dashboard.ngrok.com/signup 
# Then you can copy your authtoken found here:
# https://dashboard.ngrok.com/get-started/your-authtoken
# include the string below:
ngrok.set_auth_token("<NGROK_AUTH_TOKEN>")
```
Right now, even with the registered account, the existing meshcat code will still fail with 
```
pyngrok.exception.PyngrokNgrokError: The ngrok process errored on start: Your account is limited to 1 simultaneous ngrok client session.\nActive ngrok client sessions in region 'us':\n  - ts_23CPOyjRRv1bKnuB1mNrrS9XxMZ (35.229.148.4)\r\n\r\nERR_NGROK_108\r\n.
```
because of the default behavior I mentioned above.

russorat, 2022-01-03:
Thanks for the background!

re: free account is limited to one tunnel
The free account is limited to one ngrok agent process, but that process can open 4 tunnels. You can configure ngrok to only open a single tunnel using the `--bind-tls` flag: https://ngrok.com/docs#http-bind-tls

The error you are receiving is because you have more than one ngrok process running in your colab instance. I think running something like `!killall ngrok` before starting the agent will solve that.

RussTedrake, 2022-01-03:
Thanks!  Yes, I know the `--bind-tls`.  That's the change I need to push upstream.

I also hadn't clicked to verify my email address for the ngrok account.  That made it possible to run *one* meshcat session without changing the upstream code. So my header above should work.

Unfortunately, it seems that "factory reset runtime" on colab (which is what we have been recommending instead of something more targeted like `!killall ngrok`, because it starts students back at a clean, debuggable state) doesn't seem to immediately relinquish the ngrok agent process token.  So after a factory reset, I get the same ngrok process error on the next run.

ParthaShuvro, 2022-01-03:
Great, Thank you both. 
I ran (while being signed into my ngrok account)
```
ngrok.set_auth_token("my_token")
!pkill ngrok
http_tunnel = ngrok.connect(bind_tls=False)
!pkill ngrok

```
inside the drake installation header which seemed to worked for me. 

Thanks again.

closed 2022-01-03 by ParthaShuvro