RussTedrake/underactuated #459: cannot import name 'running_as_notebook' from 'underactuated'

littleggghost, 2022-01-24:
Hi Professor

I use colab at
```
https://colab.research.google.com/github/RussTedrake/underactuated/blob/master/intro.ipynb#scrollTo=A4QOaw_zYLfI
```
and try to run the first block, but get the following error
```
ImportError: cannot import name 'running_as_notebook' from 'underactuated' (/usr/local/lib/python3.7/dist-packages/underactuated/__init__.py)
```

RussTedrake, 2022-01-24:
Apologies.  I'll need to push a new wheel to pip; will do that today.  But this is a sign that I've transitioned to Deepnote as the primary workflow; colab support is now destined to lag a little behind.  

http://underactuated.csail.mit.edu/drake.html#notebooks

RussTedrake, 2022-01-24:
I've pushed a new wheel.  It works now.  Please remember to use `!pip3 install underactuated` at the top of the notebook.

closed 2022-01-24 by RussTedrake