RussTedrake/underactuated #497: Project guidance required

adityaanirudhk, 2022-08-15:
Greetings! I am K. Aditya Anirudh and I'm Currently pursuing my undergraduate degree in mechanical engineering. This is to bring to your kind notice about my project idea which is given below. 

Completely automated quadvehicle(kart) using a* heuristic approach/LIDAR to grid map approach/spline trajectory using De- Casteljiu's algorithm and programming of the automation system using non linear dynamics approach. 

Each of the approaches has its own cons hence multiple approaches have been framed by me have been uploaded in github. 

Components and chassis can be chosen keeping polymer matrix composites in mind which be 3D printed. *The above idea could be applied to a space rover thereby increasing its efficiency with comparatively lesser human intervention.*

The above is a rudimentary idea of my project and it would be an honor if I could accomplish this under your esteemed guidance.Could you suggest wether DRAKE software could help me in performing the non linear dynamic approach towards programming the robot and where can I find the instructions on how to use your software? A reply is awaited.

RussTedrake, 2022-08-15:
I appreciate the enthusiasm, but this github issue database isn't the right forum for asking about project advice.

closed 2022-08-15 by RussTedrake

adityaanirudhk, 2022-08-18:
I apologize for taking the liberty and commenting in this section. I had sent quite a few emails regarding my query and I hadn't heard from you for a week since the mails were sent from my side. I apologize for the inconvenience caused. The same project Idea has been sent to your mail ID. You can go through it if you would like to.

Sorry for the inconvenience caused again.

Yours faithfully
K. Aditya Anirudh