RussTedrake/underactuated #498: The Simple Pendulum.ipynb simulation not working

adityaanirudhk, 2022-08-21:
The simple pendulum, dynamics autupsy and rest not working due to missing modules names undefined. 
Following is one of the problems with simple pendulum simulation:
ModuleNotFoundError: No module named 'mpld3'

I have tried installing mpld3 module using pip installer but it says the requirement is already satisfied.

RussTedrake, 2022-08-21:
It seems that you're having some general problem with you pip installations / python paths.  It doesn't seem particular to our notebooks?

adityaanirudhk, 2022-08-22:
I have tried using !pip install mpld3 in the given jupyter nopebook to have the proper python path and then tried importing the required module, but it still didnt work. I dont understand the problem as I have confirmed that the pip installations are working fine

adityaanirudhk, 2022-08-23:
I have tried two ways to import it. One is by using !pip install command directly in the jupyter notebook, the second is by using cmd terminal and using pip install command but both of them dont seem to work. I have tried searching for the problem solution in stack overflow but i couldnt succeed. Is there any other way i could access the simulations?

RussTedrake, 2022-08-26:
The deepnote online notebooks should allow you to try everything with zero installation.

closed 2022-08-26 by RussTedrake