RussTedrake/underactuated #499: how to generate pdf correspond to latest html?

carlchan0514, 2022-11-04:
can you share the scripts for generating pdf?

RussTedrake, 2022-11-04:
https://github.com/RussTedrake/htmlbook/blob/master/make_pdf.py is the entry point.

closed 2022-11-04 by RussTedrake

carlchan0514, 2022-11-04:
i got a error "sh: 1: node: not found"

carlchan0514, 2022-11-04:
after install node and npm,i got
> puppeteer@19.2.2 postinstall /home/dongchen/learning/underactuated/node_modules/puppeteer
> node install.js

internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module 'puppeteer/internal/node/install.js'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/home/dongchen/learning/underactuated/node_modules/puppeteer/install.js:38:27)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
npm WARN notsup Unsupported engine for puppeteer@19.2.2: wanted: {"node":">=14.1.0"} (current: {"node":"10.19.0","npm":"6.14.4"})
npm WARN notsup Not compatible with your version of node/npm: puppeteer@19.2.2
npm WARN notsup Unsupported engine for puppeteer-core@19.2.2: wanted: {"node":">=14.1.0"} (current: {"node":"10.19.0","npm":"6.14.4"})
npm WARN notsup Not compatible with your version of node/npm: puppeteer-core@19.2.2
npm WARN enoent ENOENT: no such file or directory, open '/home/dongchen/learning/underactuated/package.json'
npm WARN underactuated No description
npm WARN underactuated No repository field.
npm WARN underactuated No README data
npm WARN underactuated No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! puppeteer@19.2.2 postinstall: `node install.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the puppeteer@19.2.2 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/dongchen/.npm/_logs/2022-11-04T11_10_47_404Z-debug.log

RussTedrake, 2022-11-04:
yes, you'll need to install node.js.  i'm not officially supporting people running that file.  (it would be work to support it correctly on different platforms, etc).