RussTedrake/underactuated #500: Can Only Import Tests and Grader if CWD is Root Directory (Causes Problem with Jupyter Notebooks)

cohnt, 2023-02-15:
For the exercises, I'm unable to import the tests and grader unless I manually set the current working directory to be the root directory. (Jupyter defaults to the folder containing the notebook.) Otherwise, I get a ModuleNotFoundError.

For example, with the vibrating_pendulum example, it doesn't work initially, but adding `%cd ../..` to the top (before anything is imported) fixes the problem.

RussTedrake, 2023-02-15:
You just need to add the underactuated root folder to your python path.

cohnt, 2023-02-15:
I installed underactuated with pip, so it should already be in the path from the site-packages folder. Notably, `import underactuated.exercises.pend.test_vibrating_pendulum` fails, but `import underactuated.exercises.pend.vibrating_pendulum.test_vibrating_pendulum` succeeds. So it seems like the directory structure in pip doesn't match up with the packages in this repository?

(I'm trying to avoid changing my python path variable outside of the virtual environment. But I'm okay with doing that as a temporary measure.)

RussTedrake, 2023-02-15:
oh.  yikes. that's because the underactuated pip wheel is out of date.  let me refresh that.

RussTedrake, 2023-02-15:
I've updated the wheels.  Give it a try now?

cohnt, 2023-02-15:
It works now. Thank you!

closed 2023-02-15 by cohnt