RussTedrake/underactuated #512: ImportError: cannot import name 'ConfigureParser' from 'underactuated'

zouhairm, 2023-03-20:
Hi - 
First of all, thanks for putting this course and code available online :)

I was trying to follow some of the tutorials, It seems like a few of the sample notebooks were recently updated to use ConfigureParser in this PR #509

However, the installation instructions and the most recent released underactuated package `2023.3.2` does not work with these examples:

```
$ pip3 index versions underactuated
WARNING: pip index is currently an experimental command. It may be removed/changed in a future release without prior warning.
underactuated (2023.3.2)
Available versions: 2023.3.2, 2023.2.23, 2023.2.18, 2023.2.15, 2022.6.8
  INSTALLED: 2023.3.2
  LATEST:    2023.3.2

$ python3 -c 'import underactuated; from underactuated import ConfigureParser'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: cannot import name 'ConfigureParser' from 'underactuated' (/Users/zouhair.mahboubi/projects/FlightPhysics/trimMap/.drake/lib/python3.11/site-packages/underactuated/__init__.py)
```

I did manage to get things working by installing directly from github as opposed to the released package on pypi with:
```
$ pip3 uninstall underactuated
$ pip3 install git+https://github.com/RussTedrake/underactuated.git@1292545b19299437d82dd3412b57556c13d016b6#egg=underactuated
$ python3 -c 'import underactuated; from underactuated import ConfigureParser'
$ echo $?
0
```

I imagine this will be fixed once a new underactuated release is cut, but just wanted to bring it to your attention.

Also, not sure if it's user error or the usual dependency hell, but I did struggle installing things as the instructions on http://underactuated.mit.edu/drake.html#section3 don't specify that this basically requires being on python3.11. If one just tries running `pip3 install underactuated` on anything prior to py3.11, pip will fall back to an old version of underactuated as the most recent requires `drake>=1.13.0` which doesn't appear unless I use python3.11 (despite pypi page claiming only requirement is py >= 3.8)

fazeel15, 2023-10-04:
In my case, I was getting the same error so I tried importing underactuated separately using:
`import underactuated` 
I got the following error:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/fazeel/drake/lib/python3.10/site-packages/underactuated/__init__.py", line 1, in <module>
    from .jupyter import running_as_notebook
  File "/home/fazeel/drake/lib/python3.10/site-packages/underactuated/jupyter.py", line 8, in <module>
    from ipywidgets.widgets import FloatSlider
ModuleNotFoundError: No module named 'ipywidgets'

so I installed ipywidgets using **pip install ipywidgets** and then the following lines were working for me:
```
from underactuated import (
    ConfigureParser,
    ManipulatorDynamics,
    running_as_notebook,
)
```

So I think you should import separately using `import underactuated` so see if there is any missing module.

RussTedrake, 2023-10-07:
fwiw -- i think the pip wheels are in a much better place now; and are tested on CI.  I still don't have all of the prerequisites listed properly.  Those are still in requirements.txt.  And I don't have a good way, with poetry, to specify that the required python version is 3.8 on ubuntu 20, 3.10 on ubuntu 22, and 3.11 on mac.  But I've linked to the drake docs where all of this is spelled out.  It's getting there!

closed 2023-10-07 by RussTedrake