RussTedrake/underactuated #523: No matching distribution found for drake>=1.14.0 (from underactuated)

dvogureckiy99, 2023-05-06:
### What happened?

I am install with this [Building the Python Bindings](https://drake.mit.edu/from_source.html) instructions. WITHOUT USING DOCKER CONTAINER;
And when I try `sudo pip install underactuated` I have got error:
```
ERROR: Could not find a version that satisfies the requirement drake>=1.14.0 (from underactuated) (from versions: none)
ERROR: No matching distribution found for drake>=1.14.0 (from underactuated)
```
When I wanted run `humanoids/littledog.ipynb` I have got:
```
ModuleNotFoundError: No module named 'underactuated'
```
this is the reason I tried install underactuated like above.

### Version

master branch commit 8327b29

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using CMake

RussTedrake, 2023-05-08:
I suspect that there are at least a few things happening here.
1) building drake from source then running `pip install underactuated` will still try to `pip install drake` (since your manually built will not be identified as >1.14.0).
2) since `pip install drake` seems to have failed, I suspect you're using some version of python other than 3.8, which is the [one we support](https://drake.mit.edu/installation.html) on Ubuntu 20.04?

I agree with your approach to simply `git clone underactuated`.  The step you've missed is just to add the `underactuated` directory to your PYTHONPATH.

dvogureckiy99, 2023-05-08:
Thank you very much Russ Tedrake for your work and helping.

closed 2023-05-08 by dvogureckiy99