RussTedrake/underactuated #529: Question about rimless_wheel.ipynb

dschlee, 2023-07-31:
I have a question regarding the code block that creates the phase portrait for the rimless wheel:

```python
def simulate(initial_angular_velocity=1.5, duration=1.5):
        rimless_wheel.SetDefaultContext(context)
        context.SetTime(0.0)
        if initial_angular_velocity >= 0:
            initial_angle = params.slope() - rimless_wheel.calc_alpha(params)
        else:
            initial_angle = params.slope() + rimless_wheel.calc_alpha(params)
        if initial_angular_velocity == 0:
            # Set double_support = True.
            context.get_mutable_abstract_state(0).set_value(True)

        context.SetContinuousState([initial_angle, initial_angular_velocity])

        integrator.StartDenseIntegration()
        simulator.Initialize()
        simulator.AdvanceTo(duration if running_as_notebook else 0.1)
        pp = integrator.StopDenseIntegration()

        return pp.vector_values(pp.get_segment_times())
```

Is there a way to change the function in such a way that constant angular velocity is applied to the wheel during the entire simulation? I would like to compare phase plots of the passive dynamics of the wheel on the slope vs forced angular velocity on a flat ground.

RussTedrake, 2023-08-01:
Here's a slightly long answer to a short question:

The rimless wheel dynamics are derived by hand (as opposed to using Drake's more general physics engine in MultibodyPlant).  they are in c++.  you can find them [here](https://github.com/RobotLocomotion/drake/blob/master/examples/rimless_wheel/rimless_wheel.cc).  Depending on your comfort with c++ / building Drake from source, that unfortunately makes them a little harder for you to change.

It's definitely possible to reconstruct them in python.  See the [SLIP model notebook](https://underactuated.mit.edu/simple_legs.html#running) for a similar python example.  

Porting the rimless wheel dynamics to python and adding the ankle torque to the dynamics is probably the best way. (by the way -- I do think you mean constant torque, rather than constant angular velocity... otherwise you will have canceled the dynamics and the phase portrait is trivial?).

closed 2023-08-01 by RussTedrake

dschlee, 2023-08-01:
Yeah, I meant constant torque, sorry for the mix-up. And thank you very much for the explanation!